iNavFlight/inav #2020: RFC: Prepare for GCC 7 - discussion

stronnag, 2017-08-20:
For some of us, gcc 7 is already a reality; it adds a number of new compiler warnings that we may wish to suppress. In particular, the new `switch` fall-through warnings are endemic in iNav (and its siblings). 

One option would be to globally suppress fall-through warnings through a compiler option; I'm not in favour of this as we may miss valid warnings. Another option is to add attributes or comments to the code. Adding comments will both make the intention clear and is harmless to older compilers (it's also a simple janitorial task for contributors such as I). 

In addition to the significant number of fall-through warnings, there are a few other new warnings which are probably also janitorial in nature; thus we're in pretty good shape for gcc 7 general availability. No flytime issues have been observed with gcc 7 images; it does however generate slightly larger images than gcc 6.x.

Options on the best approach are solicited. The following contrived example would generate multiple ugly warnings without the attributes / comments (which can be `//`  or `/* */` style).

````
#include <stdio.h>
#include <stdlib.h>

int main (void)
{
    int i = rand();
    switch(i)
    {
        case 0:
            i = i;
            __attribute__ ((fallthrough));
        case 1:
            i = i;
                // FALLTHRU
        case 17:
            i = i;
            /* fall-thru */
        case 42:
            i = i;
            /* Falls through. */
        default:
            printf("I'm %d\n", i);
            break;
    }
    return 0;
}
````

For the above `gcc   -Wextra falltest.c` generates no warnings, without the attributes / comments:
````
$ gcc   -Wextra falltest.c
falltest.c: In function ‘main’:
falltest.c:10:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
             i = i;
             ~~^~~
falltest.c:11:9: note: here
         case 1:
         ^~~~
falltest.c:12:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
             i = i;
             ~~^~~
falltest.c:13:9: note: here
         case 17:
         ^~~~
falltest.c:14:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
             i = i;
             ~~^~~
falltest.c:15:9: note: here
         case 42:
         ^~~~
falltest.c:16:15: warning: this statement may fall through [-Wimplicit-fallthrough=]
             i = i;
             ~~^~~
falltest.c:17:9: note: here
         default:
         ^~~~~~~
````

fiam, 2017-08-21:
@stronnag I'd go with a macro which should be added to each `case` which falls through:

```
#if __GNUC__ >= 7
#define FALLTHROUGH __attribute__ ((fallthrough)
#else
#define FALLTHROUGH
#endif
```

This way you just need to add `FALLTHROUGH;` at the end of any `case` which falls through, just like in Go. In GCC < 7 the macro will expand to an empty string and the `;` at the end will produce just an empty statement which should be harmless.

stronnag, 2017-08-21:
I would have preferred to use the simpler construct of the recognised comments, especially as a they already exist in a number of affected files. However, it seems (after some experimentatio) that `--save-temp=obj` negates the well known comment recognition, so the only solution with `--save-temp=obj` will be the conditional `__attribute__` macro.

digitalentity, 2017-09-08:
@stronnag I believe @fiam introduced FALLTHROUGH macro for compatibility with GCC 7 (used in `osd.c` so far).

stronnag, 2017-09-08:
I saw that. I'll work through the places where it's needed after 1.7.3

stronnag, 2017-09-09:
It's done in https://github.com/iNavFlight/inav/pull/2085. There's a couple of trivial legal issues needing a second opinion. I shall fly it today, weather permitting.

digitalentity, 2017-09-09:
@stronnag amazing! Closing this issue then.

closed 2017-09-09 by digitalentity