iNavFlight/inav #2021: GPS_I2C driver not working since 1.2 on CC3D

bk79, 2017-08-20:
I took out an old CC3D to  make a small 250, I flashed all the release from 1.0 to 1.7.2, from 1.0 to 1.1.17 the  i2c board was recognized, all the other the board is not recognized, configured the board with I2C_NAV and GPS enabled, am I missing some change in the configuration that now is different or did I find a bug?

digitalentity, 2017-08-21:
CC3D doesn't have enough memory to support anything but Ublox driver. You'll need to buid your own firmware.

closed 2017-08-21 by digitalentity

bk79, 2017-08-21:
how do I add it back?
I tried #define GPS_PROTO_I2C_NAV in the target.h
and io/gps_i2cnav.c \ to the target.mk and still no luck

bk79, 2017-08-21:
I think I've found the issue
inside src\main\drivers\gps_i2cnav.c
line 32- 34
#ifndef GPS_I2C_INSTANCE
#define GPS_I2C_INSTANCE I2CDEV_1
#endif

on CC3D I2CDEV_2 is the correct instance, so by adding "#define GPS_I2C_INSTANCE I2CDEV_2" to target.h makes it work.

Anyway I didn't remove any feature to cc3d just added back the driver.It can be added without any trouble, and it would help a lot a board with such a limited amount of serial ports also for airplanes attached you will find the builds I did

[inav_1.7.3_CC3D.zip](https://github.com/iNavFlight/inav/files/1238576/inav_1.7.3_CC3D.zip)