iNavFlight/inav #2022: QuadX 1 Servo setup on Aux channel (for manual tilt)

Jock384, 2017-08-21:
Hi guys, i built a 250 size quad (SPF3acro, rcsplit, fpv setup) and now im in the process of building a 500 size quad. I got My new Spf3 (deluxe) a couple of days ago and since i want to try to use a GPS so i've flashed the FC with iNav (1.7.2). Since im relatively new to the whole iNav gui/interface, syre i've been using CleanFlight a while bit Still there seem to be some differencies. I tried connecting a 9g sg90 Servo for manual tilt to pwm pin 5 on FC (while using ch_forwarding) in CleanFlight. I got the Servo to work om Aux 5 using channel forwarding but the Servo just jitters and wont turn. I'm just wondering what the easiest way to setup a standard QuadX setup

digitalentity, 2017-08-22:
Enable "CHANNEL_FORWARDING" feature. It will enable servo outputs on RC inputs 5-8. Motor outputs are not used.

Jock384, 2017-08-22:
Thx digitalentity for your quick response :) Just a couple of follow up questions, please bare with me ;) 1. Can i just enable ch_forwarding in the gui (slider in configuration)or dom i have to do it in the cli console? 2. If so, does the motor outputs (1-4) work as usual (on quadx)? 3. After i enable ch_forwarding can i then just connect my Servo to say pwm 5 and set my Aux ch 5 (rx) to a vrA (knob) for example? I run ibus protocol on a flysky fs-ia6b receiver and the fs6x radio IF that does any difference. Does the forwarding work when running ibus protocol? Last time i almost got it to work (using ch_forwarding in CleanFlight) My Servo jittered and didn't Travel correct, do i have to set Hz rate or timing in inav go get the Servo to funktion properly it Will it just work right of the bat? Sorry for all the questions, it feels lika om completely lost in inav, i was getting quite comfortable in CleanFlight following Lee@painless360 on YT :) Thanks again for your answer :D

oleost, 2017-08-22:
https://github.com/iNavFlight/inav/wiki/Connecting-a-Gimbal-to-iNav

krzysztofmatula, 2017-08-22:
I can answer as I use the channel-forwarding in my setup:
1. yes, GUI
2. this depends on the target platform. Usually there are limitations (or forwarding is not implemented at all). On SPF3 luckily the forwarding is supported. I'm not 100% sure but I think there may be 4+4 limitation (4 outputs for motors, 4 for forwarded AUX channels). Just enable the forwarding and check if which outputs are driven by the forwarded channels.
3. yes. For safety you may connect oscilloscope to the output and check if the PWM frequency is 50Hz (default value) and not anything like 400Hz or 2kHz that is used for motors.

Jock384, 2017-08-24:
Thanks for your responses guys :) I fries my FC due to a faulty connection from pdb to FC :/ i have a Spf3 acro thats flashed with inav 1.7.2 (flashed it yesterday) so im gonna try go get a servo moving tomorrow :) i connected the FC in inav succesfully and i set up the rx protocol in inav to Serial rx om uart3 and tested the tx and everything works fine. I tested enabling channel forwarding and connected a Servo to pwm pin 5 and i have my Aux 5 channel set to a knob on my tx bit nothing happens :( I don't have any motors or esc connected tough :/ i also tried servo_gimbal with servo connected to pwm 7 and 8 but still no reaction from servo :/ I do noticed that the pdb doesn't put out 5V so the fc is getting less current than 5V. I will try to rum the 5v from My 4-1 esc (bec) and see IF that does any difference ;) Any ideas if nothing happens? I just can't seem to get the Servo to respond om any pwm, also tried to run rx in ppm outputs, No difference :/ i really appreciate your answer, feeling Kinda stupid as i can't seem to get it to work nontheless :P Thx again in advance /Jocke

krzysztofmatula, 2017-08-24:
I'd suggest to first check the receiver page in the inav-configurator to see is the receiver is setup correctly and working. Also if the requested channel presents the value as sent by the transmitter.

DzikuVx, 2018-04-04:
issue appears to be dead and new mixer setup from #2978 should simplify setup only with mixer

closed 2018-04-04 by DzikuVx