iNavFlight/inav #2027: Documentation updates for next release

fiam, 2017-08-22:
Just keeping track of things that would be nice to document for the next release, either in the docs or the wiki. Note that features don't need to be recently added, just undocumented or poorly documented right now (e.g. AIRMODE has been around for a while but it seems it's not documented). Please, do comment with your suggestions. I'll update the issue as needed.

- [ ] AIRMODE: What it does in MC and FW and why it should always be enabled in FW.
- [ ] RC split support (#1977)
- [ ] i2c_speed (#1934)
- [ ] Failsafe phases in OSD (#2012)
- [ ] Targets that will be dropped after the next release. This might include all F1 targets, it seems @digitalentity is trying to keep NAZE around for the next release.

drones101, 2017-08-27:
Allowing osd and camera input to be transmit via Wifi/Bluetooth moduile to an app for android/apple.

fiam, 2017-10-05:
Marking failsafe phases as documented, since we now show a self explanatory message rather than a short string for each phase.

fiam, 2017-10-15:
I'd say that we can consider the AIRMODE part of this issue fixed. We now enable it by default on FW presets via the feature bits, so users don't need to set up modes for it.

closed 2017-10-15 by fiam