iNavFlight/inav #2029: Failsafe = RTH might not be working correctly

fiam, 2017-08-22:
A few weeks ago I had a FW model configured with `failsafe_procedure = RTH` go out of range and it failed to do return. Unfortunately, my spotter couldn't see the model and I lost video too because LoS was blocked by a hill and I didn't have an SD card inserted to record BB data, so I couldn't determine exactly why it failed to perform an RTL. At the time, I attributed it to my stupidity because I had `nav_extra_arming_safety = OFF` after running a few tests inside and forgot to turn in ON again, so I might have armed without a HOME position set.

However, a few days ago one of my flying buddies sent my this YT video https://www.youtube.com/watch?v=ahtxH5nqYy8 where the same thing happens to someone else, so I think  there might be a bug lingering somewhere. I've already asked the uploader for BB logs and I'll also be performing some tests in the next few days.

digitalentity, 2017-08-22:
Reagrding that video. You can hear motor stopping when transmitter is turned off. My guess is failsafe mis-configuration on RC/RX or FC side.

fiam, 2017-08-22:
@digitalentity Wouldn't `FAILSAFE_LANDING` without changing `failsafe_throttle` from its default `1000` value also stop the motor?

digitalentity, 2017-08-22:
@fiam that would count as "FC misconfiguration" - guy was expecting the plane to RTH

fiam, 2017-08-22:
@digitalentity I just did some tests on an OMNIBUS on FW removing the prop and running with the plane on the ground. Manually triggered RTL seems to allow me to control THR, while I wouldn't expect that. Failsafe triggered RTL (it's indeed RTL since I applied to patch to see it in the OSD) just sets THR to a low value (or maybe just keeps it at the value it was when the RC link was lost, not completely sure) and doesn't increase it. The motor is spinning, but not fast enough to keep the wing on the air. Here is my CLI diff:

# diff

# version
# INAV/OMNIBUS 1.7.2 Jul 20 2017 / 11:40:53 (81b85de)

# resources

# mixer
mixer FLYING_WING

# servo
servo 4 1000 2000 1500 -100 -1 

# servo mix

# feature
feature -RX_PPM
feature -CURRENT_METER
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper

# map
map TAER1234

# name

# serial
serial 0 32 115200 38400 0 115200
serial 1 2 115200 57600 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100
aux 1 1 1 900 1200
aux 2 2 1 1375 1675
aux 3 20 0 900 2100
aux 4 3 2 1300 1700
aux 5 9 2 1550 1700
aux 6 8 2 1800 2100
aux 7 10 5 1400 2100
aux 8 14 3 1700 2100
aux 9 27 4 1300 1625
aux 10 28 4 1800 2100
aux 11 11 6 1800 2100

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 20HZ
set acc_hardware = MPU6000
set acczero_x = 150
set acczero_y = -32
set acczero_z = -378
set accgain_x = 4119
set accgain_y = 4101
set accgain_z = 4032
set mag_hardware = NONE
set baro_hardware = BMP280
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_device = SDCARD
set failsafe_throttle_low_delay = 0
set failsafe_procedure = RTH
set align_board_pitch = 2
set small_angle = 180
set gps_sbas_mode = EGNOS
set nav_rth_climb_first = OFF
set nav_rth_allow_landing = OFF
set nav_rth_altitude = 7000
set nav_fw_bank_angle = 45
set nav_fw_loiter_radius = 3000
set nav_fw_launch_detect_time = 30
set nav_fw_launch_thr = 1800
set nav_fw_launch_idle_thr = 1200
set nav_fw_launch_motor_delay = 50
set nav_fw_launch_timeout = 10000
set nav_fw_launch_climb_angle = 23
set frsky_unit = METRIC
set frsky_vfas_precision = 1
set frsky_vfas_cell_voltage = ON
set osd_video_system = 1
set osd_units = METRIC
set osd_time_alarm = 20
set osd_alt_alarm = 120
set osd_rssi_pos = 2074
set osd_flytimer_pos = 2518
set osd_ontime_pos = 343
set osd_flymode_pos = 2508
set osd_throttle_pos = 2433
set osd_current_draw_pos = 97
set osd_mah_drawn_pos = 129
set osd_gps_speed_pos = 2273
set osd_gps_sats_pos = 2081
set osd_gps_lon = 2098
set osd_gps_lat = 2086
set osd_home_dir = 2446
set osd_home_dist = 2487
set osd_altitude_pos = 2295

# profile
profile 1

set fw_p_pitch = 20
set fw_i_pitch = 15
set fw_ff_pitch = 70
set fw_p_roll = 20
set fw_i_roll = 15
set fw_ff_roll = 30
set fw_p_level = 15
set fw_i_level = 15
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 800
set roll_rate = 35
set pitch_rate = 9
set tpa_rate = 33
set tpa_breakpoint = 1300

fiam, 2017-08-22:
@digitalentity I'll repeat the tests later today with BB, since I forgot to put the SD card in the FC. I do have the video feed recorded, but my shitty connection here would take hours to upload it.

fiam, 2017-08-22:
@digitalentity Just reviewed the footage. THR while in RTL is completely controlled by the FC, my mistake. I'll do some tests later.

digitalentity, 2017-08-22:
@fiam incorrect test.
You have `set mag_hardware = NONE`, so without GPS fix AND prior flight forward at at least 3m/s wing doesn't know it's heading and won't initiate RTH at all. 

In manual engaged RTH - it didn't really engage that's why you were able to control throttle.
In failsafe RTH - wing entered emergency landing.

fiam, 2017-08-22:
@digitalentity I'm aware of that. I was running with the wing on my hand while I had someone else handling the radio and the googles :-).

In the footage I can see that RTH failed to engage when triggered manually one of the times (the OSD displayed flight_mode = STAB) and that's what led me to believe THR could be controlled. The other two times RTL engaged properly, but THR stayed at a low value. I think this might be happening because the plane sees that it's keeping its altitude, so it sees no need to increase THR.

digitalentity, 2017-08-22:
Logs will tell us what the plane was thinking :)

digitalentity, 2017-09-08:
@fiam any news on this one?

fiam, 2017-09-08:
@digitalentity I haven't been able to test it with BB yet. Feel free to close, I'll reopen if needed when I get the chance to test it on an appropriate flying location (don't want to go into the woods to recover a plane again).

closed 2017-09-09 by digitalentity