iNavFlight/inav #2031: THR value in OSD is the one received from the RC in NAV modes

fiam, 2017-08-22:
The OSD always displays THR input by the RC. Should we change that to shown the THR applied by the FC while the user is not in control of THR? Maybe a small change in the symbol next to the THR value would be nice too.

shellixyz, 2017-08-25:
I agree it would be nice

giacomo892, 2017-09-18:
I think for the moment is better to display user applied thottle as:

1)since pilot sticks input are not recorded on BB at least if you record DVR you have the applied thottle on the video
2)I use for instance the thottle indication to center my ch3 stick when using altitude hold so I can set the copter to 0 climb/descend rate when around 1500.  I know I can simply use a bigger deadband. What about displaying climb and descend rate on OSD?  It could be a very good add!

fiam, 2017-09-18:
@giacomo892 When I say applied I mean the effective value applied by the FC. I've just realised that I didn't chose the best name, it's definitely confusing.

I hadn't realised there were some important uses of having always the RC THR displayed. I fly mostly FW with INAV and having the effective THR displayed when I'm not in direct control of the throttle definitely helps. I understand that it might be different in MC. What about providing a configuration parameter for it? We can set it to different values for MC/FW via presets.

Regarding your last point, I think you already can get climb/descent rate in the OSD. The indicator is called "Vario Num". You can also get some graphics which will show upfacing or downfacing arrows depending on the vertical speed using the "Vario" OSD indicator. We definitely need some help text for those ones.

giacomo892, 2017-09-18:
Don't worry. I'm not against the ability to show "FC computed" throttle on OSD. As you say it helps on FW on saving the plane to potential dangers and also to check how FC automatic modes behaves visually.
Maybe can't just add another element "computed throttle" and one can add both if needed.

I know OSD is becoming bigger and bigger :) But it is a very nice to have it really customizable.

fiam, 2017-09-18:
Thanks for the clarification. We definitely don't want to take away useful features while implementing new ones.

I need to think about this one a bit more. We would need 3 indicators to cover all use cases:

- **a:** always show pilot input THR
- **b:** always show effective FC THR
- **c:** show (b) in NAV modes, (a) in all other cases

Not sure if we need to cover all these use cases thought. One indicator and a setting would cover all use cases but one: having the two values visible at once.

fiam, 2017-09-20:
@digitalentity @DzikuVx Can we get your opinions on this one? I'm leaning to one of these two:

- **A)**: Leave the current indicator as is, add a new indicator which cycles between user input while user is controlling THR and effective THR while the machine is in control of THR.
- **B)**: Add a boolean setting which controls wether the THR indicator either always shows user input or whether it switches to effective THR when the FC controls THR.

Thoughts?

DzikuVx, 2017-09-22:
@fiam I'd add a new indicator switching from user throttle to auto-throttle when auto-throttle is applied

fiam, 2017-10-05:
Fixed by #2217

PR #2217 (2017-10-04): Add OSD_THROTTLE_POS_AUTO_THR indicator
Displays stick THR position when the user controls THR directly,
actual applied THR when the navigation system controls THR.

Fixes #2031

Requires updated font from https://github.com/iNavFlight/inav-configurator/pull/258 and support in the OSD tab at https://github.com/iNavFlight/inav-configurator/pull/261
commit 47fc4537 (2017-10-04): Add OSD_THROTTLE_POS_AUTO_THR indicator

Displays stick THR position when the user controls THR directly,
actual applied THR when the navigation system controls THR.

Fixes #2031
closed 2017-10-05 by fiam