iNavFlight/inav #2032: OSD_FLYMODE proposal

teckel12, 2017-08-22:
Currently, OSD_FLYMODE is:

```
char *p = "ACRO";
if (FLIGHT_MODE(PASSTHRU_MODE))
    p = "PASS";
else if (FLIGHT_MODE(FAILSAFE_MODE))
    p = "!FS!";
else if (FLIGHT_MODE(HEADFREE_MODE))
    p = "!HF!";
else if (FLIGHT_MODE(NAV_RTH_MODE))
    p = "RTL ";
else if (FLIGHT_MODE(NAV_POSHOLD_MODE))
    p = " PH ";
else if (FLIGHT_MODE(NAV_WP_MODE))
    p = " WP ";
else if (FLIGHT_MODE(NAV_ALTHOLD_MODE))
    p = " AH ";
else if (FLIGHT_MODE(ANGLE_MODE))
    p = "STAB";
else if (FLIGHT_MODE(HORIZON_MODE))
    p = "HOR ";
```

I would propose something like:

```
char *p = "ACRO";
if (FLIGHT_MODE(PASSTHRU_MODE))
	p = "PASS";
else if (FLIGHT_MODE(ANGLE_MODE))
	p = "STAB";
else if (FLIGHT_MODE(HORIZON_MODE))
	p = "HOR ";
if (FLIGHT_MODE(NAV_POSHOLD_MODE) && FLIGHT_MODE(NAV_ALTHOLD_MODE)) {
	p[2] = '3';
	p[3] = 'D';
} else if (FLIGHT_MODE(NAV_POSHOLD_MODE)) {
	p[2] = 'P';
	p[3] = 'H';
} else if (FLIGHT_MODE(NAV_ALTHOLD_MODE)) {
	p[2] = 'A';
	p[3] = 'H';
}
if (FLIGHT_MODE(FAILSAFE_MODE))
	p = "!FS!";
else if (FLIGHT_MODE(NAV_RTH_MODE))
	p = "RTH ";
else if (FLIGHT_MODE(HEADFREE_MODE))
	p = "!HF!";
else if (FLIGHT_MODE(NAV_WP_MODE))
	p = " WP ";
```

First, the order and conditions are changed to insure higher-priority modes are displayed instead of lower-priority modes.  Secondly, for position and altitude hold, it's appended to the flight mode so both are visible.  Third, 3D position hold is shown when both altitude and position hold are enabled.

So, if in angle (stability) mode and you had position hold and altitude hold active, it would display `ST3D` for stability and 3D position hold.  Previously, it would just say `STAB` in this case, so you wouldn't know you were in altitude and position hold modes.

I was thinking of just creating a branch and a pull request, but figured I'd see if there were any suggestions or resistance in doing this.

Thanks!

fiam, 2017-08-22:
I'd prefer to introduce an additional OSD element which would show the navigation modes, removing ALTHOLD, POSHOLD, RTH and WP from OSD_FLYMODE. The new indicator would use custom characters to display small icons like e.g. `[RTH]`, `[ALT-H]`, `[POS-H]`, `[A+P-H]` or `[WP]`. Using custom characters would allow to make it look pretty and use the same space without having to use the same number of characters. The disadvantage is having to maintain the character set (INAV is using ßF's right now).

teckel12, 2017-08-22:
@fiam I like that idea!  A new OSD_NAVMODE element could show multiple icons without using much space if we went the customer character route (and be more informative).  I'll look into Betaflight's custom OSD characters and extending it for nav specific characters.  This is why I proposed it for discussion instead of just submitting a pull request.

fiam, 2017-08-23:
@teckel12 Any idea of which tools are being used either by INAV or ßF to maintain the charmap? I found them in the inav-navigator project, but they're just the .mcm files.

digitalentity, 2017-08-26:
RTH, PH, WP modes are self-contained - HORIZON+PH or ACRO+PH are simply not possible. If PH fails to activate due to no GPS fix - machine will fallback to other selected flight mode (or ACRO if no other mode is configured).
The only exception is AltHold which may function alongside ACRO/ANGLE/HORIZON.

oleost, 2017-08-26:
Also althold + poshold?

digitalentity, 2017-08-26:
Technically, from code's point of view AltHold + PosHold is a 3D position hold (a separate flight mode) which implies ANGLE mode.

digitalentity, 2017-08-26:
I have another suggestion - refactor flight mode concept to disallow arbitrary flight mode combinations.

fiam, 2017-08-26:
@digitalentity I really like that. Flight mode combinations confused the hell out of me the first time I looked at INAV. Truth to be told, I was still somewhat confused until your comments on this issue because I though you could combine ALT_HOLD+POS_HOLD with ANGLE or HORIZON or ACRO on MC.

Could you please clarify which combinations are possible regarding AUTOTRIM and AUTOTUNE? From the code it looks to me like AUTOTRIM can be combined with ANGLE or HORIZON or ACRO. AUTOTUNE seems to also allow ANGLE or HORIZON or ACRO, but the wiki says that AUTOTUNE should be used with ANGLE. If that's the case, we should make AUTOTUNE switch into ANGLE too.

So if my assumptions are correct we can represent the rest of the flight modes accurately with just a line of text except for AUTOTRIM and ALT_HOLD. So I'd propose:

- Don't allow combining ALT_HOLD with POS_HOLD. Define which one should take priority over the other one and add a new 3D_HOLD mode. One possibility might be RTH > WP > 3D_HOLD > ALT_HOLD > POS_HOLD. The reason for putting ALT_HOLD with more priority than POS_HOLD is what we don't want a user enabling POS_HOLD with ALT_HOLD at the same time and making its craft drop out of the sky. This also allows us to disable POS_HOLD entirely in FW, since from my understanding POS_HOLD shouldn't be enabled without ALT_HOLD on FW.
- Clarify that all the self contained modes imply ANGLE (might be already the case?)
- Expand OSD_FLYMODE to include a small 2x1 character (top or bottom?) indicating the flight mode "options/status" active. The text will display one of the following modes (not necessarily the literal text):
    - ANGLE
    - HORIZON
    - ACRO
    - POS_HOLD
    - 3D_HOLD
    - WP
    - RTH
    - FS
    - AUTOTUNE

    - While the small indicator would show:

         - With (ANGLE|HORIZON|ACRO): `[AIR]`, `[ALT]`, `[AUTOTRIM]` or `[AIR][ALT]`. AUTOTRIM indicator could be 4x1 while the rest could be 2x1 and allow us to show them side by side.
         - With RTH and WP: Show the current state, as defined in navigationFSMState_t. Some of the states are very brief and could be omitted to simplify the code and the charmap. WP states could show a number besides them to indicate the current waypoint.
         - With FS: The failsafe phase, unless failsafe is performing and RTH. In that case, show the RTH state.

This setup has all the bells and whistles and basically fixes all complaints and issues people have with the feedback on the OSD regarding the active modes.

digitalentity, 2017-08-26:
@fiam this discussion makes me think about re-doing the mode logic alltogether. I suggest we work on 1.7.3 first and do modes and OSD stuff once it's out of the door. I intend to shuffle issues in 1.7.3 milestone to get it through RC-cycle as soon as possible.

fiam, 2017-10-05:
This is now fixed by #2201

closed 2017-10-05 by fiam