iNavFlight/inav #2035: OSD not showing good mod

voodou59, 2017-08-24:
Hi, a little bug that can be easily solved. When I activate RTH, OSD write accro mode instead and RTH works. I'm using Passthrough/Horizon/RTH. The first two are well displayed but not RTH.

Cheers.

fiam, 2017-08-24:
@voodou59 How are you testing that? It seems to me that RTH mode is not engaging because you are testing either without a valid GPS fix our without enough distance from the HOME point (where you armed the model) or maybe without a valid heading (if you're using fixed wing without MAG you need to be moving at least at 3m/s).

voodou59, 2017-08-24:
I'm 200 meters away from home, 14 satellites.
I'm waiting for a LC Filter cause video signal is not so good but once I can make a video, I'll post a link.
Moreover, if you need a translator from english to french, I'm in.

digitalentity, 2017-08-26:
Does RTH mode indeed activate? Do your plane turn home? What OSD are you using (also with what firmware)?

DzikuVx, 2017-10-06:
Due to lack of progress and no signs of actual bug, I'm closing it. Thanks

closed 2017-10-06 by DzikuVx

voodou59, 2017-10-07:
Sorry I was working. 
[image]
[image]

fiam, 2017-10-07:
@voodou59 It will never engage on the bench. Note that the boxes only turn blue when the mode is active, not when you request it to be active. In order to enable RTH, model needs to be on the armed and flying.

Note also that the channel is outside the selected range, so it would never enable RTH with that channel in the current position and that configuration.

voodou59, 2017-10-07:
Hi, ok thanks, I will try to record while flying to see it. The channel was only to show the good configuration as aux 2. When I switch it with radio it goes to RTH mode.

Thanks.

fiam, 2017-10-07:
If moving the switch on the radio puts the small marker at the bottom between the ends of the slider, it will ask the FC to engage RTH mode. Note that it doesn't mean that the mode will actually active, since there are preconditions that must be satisfied. For example, you can't enable ACRO mode until the gyro has been calibrated (you can see this in the OSD when FC is booting up, mode will show as PASS).

On the other hand, the OSD will always show active mode(s).

voodou59, 2017-10-14:
I tried it today, everything work good after a new update. Thanks for help. You can close it ;-)

boruvka112, 2020-10-18:
Hi guys,I have similar problem with iNav osd on my Matek F405 wing SE,When i switch on RTH the OSD show AIR,or if i switch to Manual it showr again AIR, I usig firmware 2,5 and Configurator 2,5,0 .The Configurator show switching dots on the mode panel on correct possition but OSD say just AIR.So i am not sure if i can go to air? Can someone give me advice please?
Thank you Bob