iNavFlight/inav #2036: Autotune crashed my wing

shellixyz, 2017-08-25:
I was trying autotune on my wing and it made it lose control. I couldn't get it to level again and it was spinning on the roll axis. I tried two times with the same result. I was flying in angle mode and did not act fast enough to manage to regain control and crashed nose first into the ground. I attached the log of the second attempt.
[LOG00011.TXT](https://github.com/iNavFlight/inav/files/1252148/LOG00011.TXT)

digitalentity, 2017-08-26:
Are you sure your wing can do 1.1 barrel rolls per second? You log specifies that you have configured max roll rate of 400 deg/s - that's a bit too much for most airplanes. Autotune is not designed to handle high rates for aerobatic flights.

shellixyz, 2017-08-26:
I'm pretty sure it can, it is fast, but it is difficult to measure without video. I was not rolling when it happened. Each times it was when I pitched hard up/down. I flew it hard since without autotune and I had no problem. Yes max roll angle is high but the problem did not happen when rolling but when pitching up/down. I attached the config.
[config_20170826.txt](https://github.com/iNavFlight/inav/files/1253934/config_20170826.txt)

shellixyz, 2017-08-26:
At 2:33.5 you can see the roll/pitch input is 0 and although I was in angle mode it was nose down and rolling.

digitalentity, 2017-08-26:
Yes, 90 deg pitch is a singularity point, where roll becomes meaningless. In that case wing can't get out of steep dive because roll transitions from -180 to +180 randomly.

shellixyz, 2017-08-27:
I understand that but I don't understand exactly what is the problem then. The max pitch angle is set to 45°. Why would it go to 90° ? The first time I tried to recover by switching to "Acro" but couldn't recover either. In any case it seems to me that autotune is very dangerous. What am I doing wrong ?

digitalentity, 2017-08-27:
You need to understand how autotune works. When you command roll/pitch autotune analyzes your command vs wing response. If wing responds too fast and exceeds the requested rate - autotune reduces PIFF gains. If wing responds too slow - autotune increses gains. In most cases your wing responded faster than expected and autotune 

Looked at the part of log you mentioned (~2:30) - it's very interesting. When you tried to pitch up the wing instead rolled right. Firmware was trying to level it back as hard as possible, however wing didn't respond.

In the earlier part of log - this is normal behavior - gyro (red) is following PID output (yellow) with a delay:
[image]

However here (at uncontrolled roll) there is no correlation between PID output and wing response:
[image]

In both cases PID output (and thus servo deflection) is of similar magnitude.

I don't think this is algorithm flaw - PID controller is working as expected, wing was not in 90deg dive but neither pitch, not roll was responding to servo commands. My best guess is a failed servo or linkage.

shellixyz, 2017-08-27:
I see. I thought about it and I realized I probably made a mistake in my servo mix. I mixed 100% for pitch and roll as you can see in my configuration when it should be 50%. Do you think it could be the cause ? I don't think it is a servo or linkage issue as they were not damaged during the crash and I flew the wing without changing anything related to the elevons and it flies just fine. It probably reacted too quickly as you said and got stuck nose down because it oscillated between 100% roll in one direction and the other and couldn't apply pitch correction. I will have to test it again after changing my mix to 50% for roll and pitch but I'm still afraid it could crash.

digitalentity, 2017-08-27:
Might be the case, however I doubt that mixer is responsible. PID controller didn't request more than 300 units servo throw - even with 100% that won't saturate the controls. I suggest that you reduce max roll angle to 50-60 deg and pitch angle to 30 deg. This should prevent edge cases. Also your bail-out mode should be passthrough, not ACRO - there you have full manual control over servos.

For initial test with 50% mix set roll D to 35-40 and pitch D to ~120-150 (according to what autotune ended up with). This is odd, however. Usually flying wing needs similar response gains on roll and pitch - might be CoG issue.

Btw, what wing is this?

shellixyz, 2017-08-27:
Yes you are right I should have switched to passthrough but I'm not sure I would have managed to recover anyway. The first time I changed mode but had 2 switch changes to go in passthrough, not very clever I know. The second time I only had 3 seconds before impact and didn't even manage to change mode even though I changed my switches to only have one to change to go in passthrough. Next time I try I will probably go higher and with FPV to give me a chance to recover.

It is an experimental wing with a custom ABS center section and foam board wings based on the FT Mini Arrow. 980mm wing span.

digitalentity, 2017-09-08:
@shellixyz can we close this issue?

shellixyz, 2017-09-14:
I am closing it. The weather is awful I can't test this for now. Will open a new one if needed. Thank you for your support.

closed 2017-09-14 by shellixyz