iNavFlight/inav #2039: Nothing since bcc30378b7faa3e56d516c9331be2eb15ea6a32d will arm

stronnag, 2017-08-26:
On my three (MATEK F405, OMNIBUS, SPEVOF3) machines, the machine will not arm on any build since bcc30378b7faa3e56d516c9331be2eb15ea6a32d

I'd missed this previously due to having only flown @fiam's branch recently.

example diff attached in case it's relevent.
[deadcat.txt](https://github.com/iNavFlight/inav/files/1253795/deadcat.txt)

stronnag, 2017-08-26:
apologies for the version confusion, 6067f2a does not arm, c30819d8 and earlier do.

stronnag, 2017-08-26:
Ah! If I go back to 095f07fe33 (i.e. revert the most recent 26c9048edbf, which coincidently bears exactly the same commit message as 6067f2a86, it magically works again).

digitalentity, 2017-08-27:
@stronnag can you try this patch:
```
diff --git a/src/main/fc/fc_core.c b/src/main/fc/fc_core.c
index 1470b144..3e156eb0 100755
--- a/src/main/fc/fc_core.c
+++ b/src/main/fc/fc_core.c
@@ -169,7 +169,7 @@ static void updateArmingStatus(void)
         }

         /* CHECK: RX signal */
-        if (failsafeIsReceivingRxData()) {
+        if (!failsafeIsReceivingRxData()) {
             ENABLE_ARMING_FLAG(ARMING_DISABLED_RC_LINK);
         }
         else {
@@ -180,7 +180,7 @@ static void updateArmingStatus(void)
         if (calculateThrottleStatus() != THROTTLE_LOW) {
             ENABLE_ARMING_FLAG(ARMING_DISABLED_THROTTLE);
         } else {
-            ENABLE_ARMING_FLAG(ARMING_DISABLED_THROTTLE);
+            DISABLE_ARMING_FLAG(ARMING_DISABLED_THROTTLE);
         }

         /* CHECK: Angle */
```

I did some stupid typos in new pre-arm checks code, my bad.

stronnag, 2017-08-27:
Closing, fixed in #2041

closed 2017-08-27 by stronnag