iNavFlight/inav #2040: New concept of flight modes

digitalentity, 2017-08-26:
Current concept of flight modes is confusing - you need to activate a combination of modes to get something working.

Suggestion - refactor concept of flight modes with following points in mind:

1. BOXes are assigned to AUX channels as usual
2. BOXes are split into "FLIGHT MODE", "MODIFIER" and "FUNCTION"
3. Only one FLIGHT MODE is active at any given time
4. (implied by the previous point) FLIGHT MODE is self-contained, no need to combine with anything else. This will also require that we add a new flight mode "ACRO"
5. MODIFIER - changes the behavior of a FLIGHT MODE. I.e. HEAD FREE could be a modifier for ANGLE mode.
6. FUNCTION is independent and used to activate certain features of flight controller. Example: BEEPER, BLACKBOX.

Navigation modes will fit nicely into this concept - they are already mostly self-contained.

mikeller, 2017-08-26:
What is the difference between 'MODIFIER' and 'FUNCTION' in terms of the actual implementation? Or rather can and should they be combined into 1, so that you have a 'FLIGHT MODE' selector, and a set of 'FUNCTION's that can be switched on and off irrespective of flight mode?

digitalentity, 2017-08-26:
@mikeller in terms of implementation - none. We'll probably have a mask to allow a FUNCTION to activate only with certain flight modes.

mikeller, 2017-08-27:
@digitalentity: Definitely. A lot of the existing modes are already dependent on some sort of condition, including a specific flight mode being active.

martinbudden, 2017-08-27:
This is definitely a welcome improvement.

fiam, 2017-08-28:
This is definitely an improvement over the current system. Any mode which only works with ANGLE and is unlikely to work with HORIZON/ACRO in the future should be "promoted" to "FLIGHT MODE".

Now, I'm thinking about how to show this on the OSD. How many MODIFIER modes which can be applied to at least two of ANGLE/HORIZON/ACRO do we have? I guess AIR could be applied to any of them, but what about HEADFREE, AUTOTUNE and AUTOTRIM? I'd really like to be able to show all of this to the user without requiring to set up a complicated OSD configuration (probably by making OSD_FLYMODE two lines).

Also, the more I think about it, the more I'd like an OSD element for temporary messages. It would allow us to show when a function mode has been enabled/disabled (e.g. "Beeper is on", "Beeper is off") or the reason for not arming (e.g. "Gyro not calibrated") or the current RTH phase (e.g. "Climbing", "Going home", "Landing").

Bobrcnut, 2017-08-30:
How (if at all) will this affect the LEDstrip function?  To quote fiam:  "I'd really like to be able to show all of this to the user without requiring to set up a complicated OSD configuration".  LEDstrip usability has been degraded over time because of the differences between INAV and its predecessors.  In my opinion, LEDstrip should offer practical function rather than 'bling'.  Some documented LEDstrip functions no longer work (or are very difficult to figure out).. it would be great to have an LED-displayed indication of individual flight modes based on color/blinking of individual LEDs.  Currently, there is no way that I can find to display some of INAV's modes/modifiers/functions.  I have no particular need for 'cylon' or 'ring' or 'turn signals' because, as an INAV fan, I'd rather have practical LED functions than eye-pleasing but functionally useless options.
     If I'm too far off topic here, let me know and I'll open a separate issue.  I'm new to all this GitHub stuff.

mikeller, 2017-08-30:
@Bobrcnut: Your comment is off topic for this issue. Please open a new issue with your suggestion, so that we can keep track of it.

kabturek, 2017-09-24:
> Also, the more I think about it, the more I'd like an OSD element for temporary messages. It would allow us to show when a function mode has been enabled/disabled (e.g. "Beeper is on", "Beeper is [...]
This is something i'm missing from other flight controllers (notably vector).

sincoder, 2017-10-15:
i think now concept of flight modes is the best, just add a extra check of the combination is enough

digitalentity, 2017-10-15:
@sincoder "the most flexible" - yes, "the best" - I'm not so sure. There are too many combinations of incompatible modes to check - and checks happen in a lot of different places in the code. It's very easy to miss something when adding a new mode - and then have a report about something bad happening because used enabled the two together.

sincoder, 2017-10-15:
maybe we can learn how DJI  does, flight mode P , A ,  F , S , function button : RTH , STOP ,Intelligent flight mode: head less ....etc, just as you think

closed 2018-05-27 by stale[bot]