iNavFlight/inav #2043: RTH not returning home, circling instead

Dejw4g, 2017-08-27:
Hi,

look at this video that I made:

https://www.youtube.com/watch?v=WxFjGKOZaoU

I tried to switch on RTH two times in this video and the airplane started circling, losing height and over-using throttle. I tried to change the RTH mode but with no change, still doing the same.

This behavior started with iNav 1.7, when I was using 1.6 everything was OK, it just turned back the plane and flew home

here is my dump:
`# version
# INAV/SPRACINGF3 1.7.1 Jun  3 2017 / 08:13:23 (48e1feff)

# resources

# mixer
mixer AIRPLANE

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 -100 -1 
servo 3 1000 2000 1500 -100 -1 
servo 4 1000 2000 1500 -100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature RSSI_ADC
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

[... 412 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 10
set pitch_rate = 10
set yaw_rate = 5
set tpa_rate = 0
set tpa_breakpoint = 1500`

fiam, 2017-08-27:
@Dejw4g Thanks for the report. Could you upload your Blackbox logs so we can see what the FC is doing?

Dejw4g, 2017-08-27:
@fiam Unfortunatelly, I don't have any, but I will go to fly it again to get one

fiam, 2017-08-27:
@Dejw4g Thanks!

Redshifft, 2017-08-28:
@Dejw4g  for a start you have it turning too hard, it is almost a spiral dive.

I would reload put everything to default and understand the settings and values, then work them up 1 at a time, observing their effect on your plane. something is not right for sure.

Tdogb, 2017-08-28:
This is not a bug. You need to allow landing in the cli. Run "set nav_rth_allow_landing = ON" Should fix your problem

Dejw4g, 2017-08-28:
@Tdogb thank you, will try, but just a question - how does this setting affect the behavior mid-air? I mean why it was circling when it was like 600m away from "home"?

Tdogb, 2017-08-28:
There is a known issue with your version: "RTH descent might not work at all, be too slow or too aggressive and don't respect the landing speed on a fixed wing" I would either update or go back to 1.6

fiam, 2017-08-28:
@Tdogb Are you sure about that? I believe to me the problem with FW landing on 1.7.1 was that it dove for the ground too fast, it shouldn't cause the problem shown in the video. @digitalentity, could you take a look at this one?

Tdogb, 2017-08-28:
@fiam It seems like there might be some funny business with rth in 1.7. I am sticking to 1.6, seems like 1.7 is too buggy at the moment (I can't really afford bugs, since I am flying a 2m wingspan skyhunter)

digitalentity, 2017-09-05:
@Dejw4g the issue in the video looks like a stall when airplane banked too hard and lost lift. Without a blackbox log it's impossible to diagnose accurately.

@Tdogb RTH logic itself didn't change between 1.6 and 1.7.x. 1.7.2 improved (fixed) RTH landing and fixed some bugs that may cause disarm mid-air when going into failsafe. I'd say 1.7.x is safer to fly than 1.6.

digitalentity, 2017-09-05:
I can't confirm the behavior neither on my flying wings nor on a classic airplane - RTH is working properly, so I doubt it's a algorithm issue - more like a setup or tuning issue.

Dejw4g, 2017-09-05:
@digitalentity should I change this value nav_fw_bank_angle to 20 instead of 45 then?

digitalentity, 2017-09-05:
@Dejw4g yes, I use 30 deg on my flying wings. They can fly stable up to 60 deg, but for automated modes I prefer to stay safe.

DzikuVx, 2017-10-06:
Since issue is stuck, I'm closing it. Thanks

closed 2017-10-06 by DzikuVx