iNavFlight/inav #2047: Enabling I2C2 on Matek F405-OSD

aster94, 2017-08-28:
Hello, 

Right now it is "trivial" to use the i2c on the F405-OSD since you need to use the i2c3 ( on S4 and S6. My proposal would be to use the i2c2 instead. 

As you know far away better than me the stm32 chip are capable of alternate funcions on the GPIO (known as AFIO). We can see on datasheet of the STM32F405RGxx that the pins for the i2c2 are shared with USART3 (which throught hole pads are perfectly exposed on the F405-OSD) 

I really think it could be a good improvement with the Matek board to make all more "plug&play" 

See rcgroup discussion: https://www.rcgroups.com/forums/showpost.php?p=38183690&postcount=15399 especially #15402 and #15403 where stronnag suggested to create two different target for the AIO and OSD

stronnag, 2017-08-28:
Ideally, we would have one target, but I'm not aware of how we would define which I2C bus to use at run time.  Uses of the AIO who have build a machine using the vendor documented dedicated I2C pins and USART3 as a serial would be less than content with a unified target that moved the pins. 

Given the scarcity of exposed outputs on these boards, and the variety of machine configurations that iNav supports, using S4/S6 neither trivial nor a realistic option.  

I agree with @aster94 that taking over USART3 is the best solution for the OSD variant; the question is how to do so in the least disruptive manner.

stronnag, 2017-08-28:
So you could  clone the experimental fork https://github.com/stronnag/inav-jh.git and then checkout the jh-matek-f405-osd-i2c branch and see if works with I2C on USART3. Experimental hex http://seyrsnys.myzen.co.uk/MATEKF405OSD/inav_1.7.3_MATEKF405OSD.hex.gz

No doubt by the time I see any replies, @digitalentity will have told us how to mange this properly :-)

aster94, 2017-08-29:
Ok really i am ordering the board only now, so for the test i can t help since it will arrive in about 3 weeks...

I just had a look on your forked repo, in src/main/drivers seems that all the work is already been done, all the possibles i2c devices are already setted. 
So i went to targets and even there there are already two folders where the AIOhas only i2c1 while the OSD has both 1 and 2

I don t know enough how the iNav works to try to make any kind of change but someone who have experience with it should be able to make it working (?)

digitalentity, 2017-08-29:
Ideally we should have one target, but I don't think it's possible at current state of development.

jabonloco, 2017-09-02:
Hi. Any news news this?

stronnag, 2017-09-02:
I'm hoping someone will test it ....

jabonloco, 2017-09-02:
I'll do It. I'm just waiting some parts.

realclicker, 2017-09-02:
I can test it tomorrow.
So I flash the hex from above and connect sda/scl to uart3 tx/rx pads?
Is that correct?

stronnag, 2017-09-02:
That is correct. I don't have the board, and I'm not very confident at this low level, so I have c. zero confidence it will work, but we'll see soon :-).

aster94, 2017-09-02:
stronnag could you show me where and how in the code did you activated the AFIO? i have a few stm32 boards around and i can write a simple test program to see if pb10 and pb11 are serial3 or i2c2

stronnag, 2017-09-02:
I cannot. The current header file is simply an assignment of I2C function to pins. If more is required, then it requires more expertise than I can offer.

aster94, 2017-09-02:
i am not sure that it will work
i will try to give you the code for the AFIO (i am an hobbist so if someone read this and knows how to do it he is welcomed to write the correct code)
it could be insert in the "setup" or anywhere you are sure the code will run only once

aster94, 2017-09-02:
maybe there is no need to no anything:
from https://github.com/iNavFlight/inav/blob/master/src/main/fc/fc_init.c line 409

```
#ifdef USE_I2C
#if defined(I2C_DEVICE)
    #if defined(I2C_DEVICE_SHARES_UART3)//we should activate it
        if (!doesConfigurationUsePort(SERIAL_PORT_USART3)) {//and deactivate this
            i2cInit(I2C_DEVICE);
        }
    #else
        i2cInit(I2C_DEVICE);
    #endif
#endif

#if defined(I2C_DEVICE_EXT)
    #if defined(I2C_DEVICE_EXT_SHARES_UART3)
        if (!doesConfigurationUsePort(SERIAL_PORT_USART3)) {
            i2cInit(I2C_DEVICE_EXT);
        }
    #else
        i2cInit(I2C_DEVICE_EXT);
    #endif
#endif
#endif
```

it is needed a little bit of dig inside this or better someone who wrote the software cames here

realclicker, 2017-09-03:
I've tested it with two BMP280 modules working on the Matek AIO, but unfortunately not on the OSD variant. :(

digitalentity, 2017-09-07:
On Matek405 USART3 is mapped to PC10/PC11 - these pins are not shared with I2C2 (which is on PB10/PB11). I don't see a viable solution for enabling I2C on MatekF405_OSD except going with emulated bit-banged I2C. Driver is in the code, but it needs some work on bit timing.

aster94, 2017-09-07:
Mmm that s bad, i thought serial3 was mapped to the default pins, i still doesn t have my board so i can t check where the throught hole are really connected on the stm32, i will email matek's support to have a chart of the board

aster94, 2017-09-07:
Bad news, they said that s4 and s6 are the only IIC pins availables
I m pretty sure we could close

aster94, 2017-09-07:
Maybe if digitalentity or whoever knows how it works have time (or explain me how to do it) we could add the soft IIC

digitalentity, 2017-09-08:
@aster94 I'll add fixes to SoftI2C drivers for better performance and create a test build for MATEK405 with shared I2C/UART3. What do you think?

aster94, 2017-09-08:
That would be really great, thanks digitalentity :D

realclicker, 2017-09-08:
@digitalentity 
Yes, please do it. :) I am here for test it out immediately.

SimplyKot, 2017-09-10:
I just completed test. It's working.
Entering CLI Mode, type 'exit' to return, or 'help'
```

# version
# INAV/MATEKF405OSD 1.7.4 Sep 11 2017 / 00:06:35 (bdc55fa)

# bootlog
Time Evt            Description  Parameters
   2:  0          CONFIG_LOADED 
   2:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 502:  2          PWM_INIT_DONE 
 502:  3       EXTRA_BOOT_DELAY 
1188:  9         GYRO_DETECTION  (7, 0, 0, 0)
1779: 10          ACC_DETECTION  (8, 0, 0, 0)
1794: 11         BARO_DETECTION  (3, 0, 0, 0)
1794: 20        PITOT_DETECTION  (0, 0, 0, 0)
1794: 12          MAG_DETECTION  (2, 0, 0, 0)
3475:  4       SENSOR_INIT_DONE 
5182:  8           SYSTEM_READY 
```

digitalentity, 2017-09-11:
Fixed

closed 2017-09-11 by digitalentity

davidbio110, 2017-09-20:
How to connect GPS and barometer to matek f405 aio fly controller,  I don't no about i2c and What is SDL and SCL, how to use 1.7.3 INAV Version. Please upload your video to YouTube
Please consider my question, this is my first drone 
Please help me .

stronnag, 2017-09-20:
This is not a support channel, it is for bug reporting and collaborative development. Please use RC Groups for support.

* You need the development branch for I2C on Matek F405 OSD
* See https://github.com/iNavFlight/inav/pull/2097 for details, including a working 1.7.4 hex file
* See https://github.com/iNavFlight/inav/blob/development/docs/Board%20-%20MatekF405.md for development branch board documentation

davidbio110, 2017-09-21:
Matek f405 AIO FLY CONTROLLER  HAVE
SDA AND SCL

   Board%20-%20MatekF405.md for development branch board documentation

or mute
the thread

aster94, 2017-09-21:
David this is for the F405-OSD not AIO and as they already said you it is better to write on the rcgroup forum please help to keep github clean

KnuckleUpFPV, 2018-01-25:
Was this resolved? Are the i2c working on uart3. Fellow on inav facebook is having issues. He's connected to uart3. Is the i2c still on the s4 s6 pads?

stronnag, 2018-01-26:
Please see the board documentatio; on the F405 OSD, soft I2c is used via the UART3 pads, on the F405 AIO there are dedicated I2C pads. Both work perfectly. With the newer STD board, you need the development branch.

KnuckleUpFPV, 2018-01-26:
Read the docs. Thats the only reason I asked here if it was resolved. It wasn't clear. I read the entire rc group thread. Also a waste of time. It's kind of stagnated. I saw something about the development branch but 1.8 is the newest. I was trying to figure out if it was working in 1.8 couple different places point you in different directions without a clear answer. I found an issue where this question was asked directly. The response was dig through rcgroup nonsense or come here. All 3 of them leading to the same conclusion. 

Thank you for letting me know we still need the development branch. I hope it helps someome else and saves them time. 

(Use the development branch as of 26 Jan 2018)