iNavFlight/inav #2054: Uploading OSD font over MSPv2 hangs the board

fiam, 2017-08-29:
Steps to reproduce:

- FC firmware from development
- inav-configurator from master
- OSD tab, click upload font and it will start getting timeouts for MSP messages
- Restarting the configurator and clicking on "Connect" will not connect to the FC

Haven't had time to debug if the problem is on the configurator or the FC side, but the apparent hang up of the FC seems to indicate the problem is on its side.

fiam, 2017-08-29:
Forgot to mention that disabling MSPv2 in the configurator allows uploading a font, so it must be MSPv2 related.

digitalentity, 2017-08-29:
I'm on it

digitalentity, 2017-08-29:
Can't reproduce

fiam, 2017-08-29:
@digitalentity I can reproduce it on OMNIBUS (F3). Are there any read alignment issues which could affect F3 and not F4?

digitalentity, 2017-08-29:
@fiam KFC32F3_INAV target (F3 CPU) - no issues.

digitalentity, 2017-08-29:
Are you sure you are using latest configurator master?

fiam, 2017-08-29:
@digitalentity Just checked twice that I was using the development branch of INAV and the master branch of the configurator. It definitely hangs. Check out this screenshot:

[image]

Unfortunately, I won't able to properly debug this until next week. Feel free to assign it to me and put on hold until then.

digitalentity, 2017-08-29:
Hm... either this is local to OMNIBUSF3 target, or we have yet another case of heisenbug.

digitalentity, 2017-08-29:
I'll send it your way, but try to debug in the meanwhile. Unfortunately I don't have an F3 Omnibus board here :(

digitalentity, 2017-08-29:
Found an Omnibus board in a broken tricopter (that is waiting to get disassembled and thrown away). Still can't reproduce the bug. It's something on your end.

fiam, 2017-08-29:
@digitalentity My apologies for the wasted time. I'll try to find some time to debug it properly as soon as possible.

digitalentity, 2017-08-29:
@fiam no worries. There still might be a bug lurking somewhere. There is a reason why it's not working for you and thus it may not work for somebody else.

fiam, 2017-09-01:
Just did a few more tests. Changing `MAX_NVM_FONT_CHAR_SIZE` to any value other than 54 works and uploads the font over MSPv2 without any problems. Lowering the value requires changing the FC to not read past the input buffer, but sending more bytes that needed doesn't hurt since MSP handles it just fine (the extra spaces in the character are 85 != '$', so MSP keeps waiting for '$').

Sending 55 bytes per character and changing the FC to read 55 bytes works too. That means that there are some shenanigans happening at 54 bytes, which results on a full MSPv2 packet of 64 bytes (54 font + 1 addr + 9 protocol). That suggests to me that there might be an off-by-one lingering somewhere, since 64 is 2**6 and a plausible size for a buffer somewhere.

I've also found that the FC is not hanging, since it responds to commands from the RX, but it no longer responds to MSP commands, even writing directly to its serial port from another process.

Next step is hand-crafting some 64 byte packets using both MSPv1 and MSPv2 and see what happens.

fiam, 2017-09-01:
Just did 64 byte packet tests. Sending a 64 bytes `MSP_API_VERSION` with a payload full a zeros blocks further communication with the FC, with both MSPv1 and MSPv2. Sending it with either 63 or 65 bytes doesn't cause any problems. Note that I'm sending those packets writing directly to the serial device, so the issue can't be in the configurator.

The issue also happens on 1.7.2. A 64 byte packet will block further communication. We either have a problem at the serial layer or there's a bug in the computer side on macOS (e.g. the VCP driver), which seems like the most likely culprit. I've already updated the driver to the latest version (released in August this year) and, unfortunately, the problem persists so we'll probably need to find a workaround.

Can someone run this program a few times against an OMNIBUS F3 on Windows or Linux and let me know if it prints the response from the board in all runs? You'll probably need to adjust the path for the serial port, then just start it with `go run msp.go`.

https://gist.github.com/fiam/f9a9cd2f992558408650b562b5706848

If it works fine on Windows or Linux, then the problem is definitely on the driver. What a pain to debug this one!

digitalentity, 2017-09-01:
@fiam why it should be Go and not Ruby, Perl or C which I already have on my system??? was a rethorical question, sorry :rofl:

digitalentity, 2017-09-01:
@fiam tested against SPRacingF3, REVO (F4) and SPRacingF4. No issues, `msp.go` reports a valid MSP response, no issues connecting to the board afterwards.

fiam, 2017-09-01:
@digitalentity Thanks for the tests! Used Go for convenience. Ruby and C would be a pain to deal with the low level stuff for this case, while Go has ready to use libraries to access the serial ports.

Since it works fine in other platforms, I guess we should wait for someone else to confirm this issue on their macOS system and once it's clear the problem is not specific to my setup, add a workaround. It can be as simple as appending a zero at the end of a buffer if its length is 64, MSP should deal with that without problem.

digitalentity, 2017-09-01:
@fiam adding a garbage byte is not a good practice, but if we have no choice - I'd rather send garbage then block the communication :)

fiam, 2017-09-01:
@digitalentity I agree, it's not ideal. Unfortunately fixing the macOS VCP driver, which seems to be the most likely culprit unless it's something on my system, is probably not an option.

Forgot to mention that I was able to also test on OMNIBUSF4 and I can reproduce the problem with it too.

fiam, 2017-09-01:
@digitalentity This is the issue that keeps on giving. I can reproduce it in an OMNIBUSF3, but not anymore on an OMNIBUSF4. In fact, I'm not quite sure it ever happened on the F4 since my test program wasn't doing as much probing as it does now.

I think we should wait until 1.7.3 is released and keep this around until we get feedback from users. Since 1.7.3 will include font changes, we'll know soon enough if anyone is having problems uploading the fonts.

stronnag, 2017-09-03:
Inspired by this, mwptools contains an evil msp fuzzer.
````
$ ./mspfuzz  --help
Usage:
  mspfuzz [OPTION?]  - msp tester

Help Options:
  -h, --help       Show help options

Application Options:
  -b, --baud       baud rate
  -d, --device     device
  -c, --cmdmax     maximum comamnd value generated [4095]
  -p, --paymax     maximum payload size generated [4095]
  -e, --evil       degree of evilness (default=no dangerous commands, 1=no reboot, 2=no reboot or dataflash commands
  -l, --loops      number of loops (default no limit)
````
For command / payload ranges in the MSP V1 range, the resilience of the FC to abusive behaviour is remarkably good. For example, with an F1 (flip32 / naze) as the victim:
````
$ ./mspfuzz -c 254 -p 192 -e 2 -l 50000
````

* 50000 iterations, command and payload in the V1 range, no reboot or dataflash commands.
* Sending dataflash read commands with random parameters (on a device without flash) can lock up the FC --- not really surprising.

There were 255 examples of message size == 64 bytes. On Linux x64, the test completes successfully:
````
801s, rx 532371b, tx 5077266b, (665b/s, 6340b/s) messages 50000
````
Currently, it appears we have limited (same as V1) provision for MSPV2 input; command is treated as `uint8_t` and the maximum payload remains 192 bytes. OTOH, at the moment, there is no requirement for anything more.

digitalentity, 2017-09-04:
@fiam @stronnag ok, seems this is not a firmware issue. Going to move to 2.0 milestone so we'll be aware of it.

digitalentity, 2017-09-04:
@stronnag regarding max 192 byte payload - it's a limit for targets without dataflash. For some MSP commands we already have > 256 byte payloads so we should probably bump up minimal buffer size to 512 bytes.

fiam, 2017-09-04:
@digitalentity Agree. Let's keep it open to not forget about it, but it shouldn't be a priority right now.

fiam, 2017-09-10:
Unfortunately, #2104 didn't fix this one. Problem still persists.

stronnag, 2017-09-10:
OTOH, this problem does not exist, at least when the host is Linux.
````
# max payload size 192, max command value 254, 10240 random messages (NAZE:) target
mspfuzz -p 192 -c 254 -l 10240 | tee /tmp/m.log
...
178s, rx 125118b, tx 1053192b, (703b/s, 5916b/s) messages 10240
$ grep mlen=64 /tmp/m.log  | wc -l
52
$ echo  echo $((52 * 192 ))
9984
````
Seems to show that rand() is quite random rather than a firmware issue? Or am I missing something?

fiam, 2017-09-19:
Just isolated the problem. Is definitely macOS specific. Using exactly the same board I can reproduce always reproduce it on macOS, while sending 64 byte MSP messages works fine on Windows.

krzysztofmatula, 2017-09-19:
The issue is not necessarily related to MSPv2. I'm having issues with blackbox log downloading on mac since 1.7.3 (SPRACINGF3). I tried to force MSPv1 and it didn't help for that case. If we want to have workaround for mac we'd need to have a way to also disable the jumbo frames on log transfer (I bet it is also a the what makes problems on mac), as well, as the MSPv2...

etracer65, 2017-09-20:
Based on previous troubleshooting with Betaflight configurator and blackbox downloads I don't think this is a VCP or Silabs driver issue, but rather a Chrome serial handling bug.  I was able to replicate the serial corruption with both VCP devices and CP210x.  To further test this I wrote a standalone blackbox download application completely outside of Chrome and could not replicate the serial corruption issue.

https://github.com/betaflight/betaflight-configurator/issues/435

The workaround to resolve blackbox download issues was to implement a MSP packet retry when CRC errors were detected.  https://github.com/betaflight/betaflight-configurator/pull/424

Other related issues in Betaflight configurator:
https://github.com/betaflight/betaflight-configurator/pull/417
https://github.com/betaflight/betaflight-configurator/issues/411

digitalentity, 2017-09-20:
We were able to reproduce the issue on MacOS using a test program in Go.

krzysztofmatula, 2017-09-20:
Do you think that 64-byte long packets also happen while downloading the log?
It stalls for me at different moments. Usually somewhere between 100kB and 1MB. So, not at very beginning, nor at very end. In fact it never reaches the end for me now.

During the download process I see the packet error counter increased (to >100 sometimes), so it's not that first crc error breaks the process. I briefly compared the js/msp.js between betaflight and inav and they are totally different... I bet some error resistance is there in the inav's version, but maybe it is not enough...

closed 2022-09-03 by DzikuVx