iNavFlight/inav #2071: SP Racing F3 Neo Tilt Gimbal

gfacek, 2017-09-03:
Hi,

I can't use the servo gimbal.

If I understood it right, usually tilt and roll server have to be connected to PWM8 (servo 0) and PWM7 (servo 1). But the SP Racing F3 Neo only have 6 motor outputs.

So, on my quad PWM1-4 are the 4 motors. I connected the tilt servo to PWM5. When I use ONESHOT125 the servo is giving some noise. When I configure the protocol to STANDARD and refresh rate to 50Hz, the noise is gone, but the servo forces to turn in one direction.
It seems, that the missing motor outputs are not considered right.

Is this possible, or am I doing something wrong?

Kind regards
Tobias

digitalentity, 2017-09-08:
Please provide details of your setup - CLI dump output is a must to troubleshoot.

DzikuVx, 2018-05-09:
I'm affraid SPRacingF3 NEO does not have any outputs that are usable in multirotor configuration. It will not work

closed 2018-05-09 by DzikuVx