iNavFlight/inav #2072: servos moves with throtle

ksakwa, 2017-09-03:
HI. I have a big problem.
In my CC3d Atom with Inav 1.7.2 (also betaflight), when stabilisation is ON and i  increase the throttle, the servos (ail and elev move to the right or left). When TH is 0, the servo returns to neutral. can you help me ?
link to youtube:
https://youtu.be/GSfwF1dDH7E

digitalentity, 2017-09-04:
Please provide more details about your setup. CLI `dump` is a must to troubleshoot.

cabinw, 2017-09-06:
@ksakwa Your fc might not be leveled and inav is work hard to let your plane be leveled but it didn't work so the servo kept moving. I guess

fiam, 2017-09-06:
If you haven't enabled AIR mode, it seems to be working as intended. The FC will not try to level itself when THR is zero. As soon as you apply THR, the FC starts applying stabilisation because you have it tilted on the table. Have you enabled AIR mode?

DzikuVx, 2017-09-26:
Since no new data appeared, closing this one. In case of new data please reopen

closed 2017-09-26 by DzikuVx