iNavFlight/inav #2075: Failsafe issue with Fixedwing on Development branch

oleost, 2017-09-04:
When I try and bench test failsafe it just seems to be stuck in an idle state.

I have not gotten a chance to test GPS hold / RTH in air yet after reconfiguration.

DIFF: https://pastebin.com/avnsLw9R
Board SPRACINGF3MINI

[LOG00177.TXT](https://github.com/iNavFlight/inav/files/1274829/LOG00177.TXT)
First log: Wait for GPS-> ARM and apply throttle -> Run approx 15 meters -> Turn of TX.
Airplane stuck in idle phase.

[LOG00178.TXT](https://github.com/iNavFlight/inav/files/1274830/LOG00178.TXT)
Second log: Wait for GPS-> ARM and apply throttle -> Run approx 15 meters -> **Enable RTH**
Airplane disarms / Cuts throttle.

oleost, 2017-09-04:
https://youtu.be/EwQMRKjUPyc

digitalentity, 2017-09-04:
@oleost there is a caveat with detecting if home is reached on airplanes. In case of RTH `waypoint_radius` is ignored and `1.1 * loiter_radius` is used instead - to allow detecting that plane reached home while it's loitering around.

So, for LOG177 place successfully passed through ALL stages of RTH, detected that it reached home due to being within 55m radius (with default `loiter_radius`), started landing and entered gliding phase due to being below 5m of altitude.

For LOG178 - same thing.

oleost, 2017-09-04:
@digitalentity Regards to LOG177, I know this is emergency and not to be used on a daily basis, but shouldnt motor been cut ? Not much help in idling at that RPM, and it would have destroyed something going like that on ground

digitalentity, 2017-09-04:
@oleost LOG177 is failsafe. MOTOR_STOP is forcibly disabled when in failsafe state to prevent zero throttle set on receiver from causing a disaster.

oleost, 2017-09-05:
@digitalentity I dont see why it should matter ?

Of course failsafe must work regardless what throttle the receiver sends out.

But when it is in failsafe, the FC must command appropriate regardless of throttle value from receiver. Whether it is cruise throttle in air, or zero throttle when it believe it has landed?

digitalentity, 2017-09-05:
@oleost we have the issue to track MOTOR-STOP and navigation modes #1940

digitalentity, 2017-09-08:
@oleost can we close this?

oleost, 2017-09-08:
Has closed it, haven't have time to retest walking 50m away yet.

closed 2017-09-08 by oleost

digitalentity, 2017-09-08:
Thanks. Please re-open if you find other failsafe-related bugs

fiam, 2017-10-05:
@oleost Did you manage to run the tests again at a bigger distance?

oleost, 2017-10-05:
@fiam No I believe I haven't gotten the chance to test yet.