iNavFlight/inav #2077: SoftSerial GPS not working.

efrenbg1, 2017-09-04:
Hi, I'm reconstructing my Quad after a year or so. I noticed that for good reasons they dropped support for the GPS on Cleanflight. So I decided to try INav insted. When configuring my ports, I have IBUS on the only Hardware Serial (Naze32). Some guides on Internet explain how to connect the GPS via SoftwareSerial, but after connecting it to my board, there is no way to make it work. The same configuration works fine of Hardware Serial, but on Software it does not even say it is been detected. My GPS is an u-blox M8N working at a baudrate of 9600.

Thanks.

digitalentity, 2017-09-04:
Please provide a CLI `dump` of your configuration.

efrenbg1, 2017-09-04:
# INAV/NAZE 1.7.2 Jul 20 2017 / 11:35:11 (81b85de)
mixer QUADX

mmix reset

servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 
smix reset

feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature PWM_OUTPUT_ENABLE
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
beeper LAUNCH_MODE
map AETR1234
name -
serial 0 1 115200 38400 0 115200
serial 1 0 115200 38400 0 115200
[... 264 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

digitalentity, 2017-09-05:
Try settng `set gps_auto_baud = OFF`. Automatic baud detection tries high baud rates which may cause softserial hickups (it doesn't like speeds more than 19200 on Naze32).

Also, be aware that 9600 baud is too low for reliable GPS operations.

DzikuVx, 2017-09-26:
Since no new data appeared, closing this one. In case of new data please reopen

closed 2017-09-26 by DzikuVx

Stalx75, 2018-03-11:
Hi. I joined your community later than the problem.
I have exactly the same situation as efrenbg1.
I have FC NAZE 32 full rev6, PPM-RX , MWOSD in UART1, on UART2 installed telemetry 3DR mavlink, activated SoftSerial, and I'm trying to use GPS. in the [Ports] tab everything is installed, everything is turned on in the "configuration" tab, but there are no changes: the INAV configurator displayed the "GPS" icon in red, there is not a single received packet in the [GPS] insertion. I tried to listen to the monitor COM port SS1 & SS2 - the data from the receiver go, but INAV persistently does not see them. I tried to transfer the telemetry to the softSerial (with the corresponding changes in speed and settings) again listened to the monitor and realized that the softserial RX and TX is not allo ... Tell me, please, how to make the GPS work on the softSerial. Thank you

stronnag, 2018-03-11:
GPS will not work reliably on softserial, period.
And we no longer support F1 boards.

expipiplus1, 2019-07-06:
@stronnag is that the case for F1 boards, or in general? If the latter then it may be worth correcting the documentation here: https://github.com/iNavFlight/inav/wiki/GPS--and-Compass-setup#getting-started-with-ublox-gps

stronnag, 2019-07-06:
I would not trust GPS for **navigation** on soft serial, period. If you just want to know where you are / have been / iNav radar it's probably OK on F4 and later.

expipiplus1, 2019-07-06:
Thanks for the response @stronnag, would you be able to expand on why? Is it a reliability concern?

stronnag, 2019-07-06:
Reliability. You will lose data.