iNavFlight/inav #2078: Add combined ONTIME/FLYTIME OSD indicator and make it the default

fiam, 2017-09-04:
When it comes to time indicators, I mostly care about "fly time" since "on time" is not very useful while flying and takes too much space on the screen for what it provides. Since I switched from MWOSD (which has combined time indicator) to ßF/INAV's OSD (which has 2 indicators for time), I've stopped using "on time".

However, until you're on the air, the "fly time" indicator is useless and just displaying all zeroes. That's why I feel the dual "on time" while unarmed / "fly time" while armed provides the best compromise between screen real state and useful information.

I believe that adding a dual "fly time" / "on time" indicator as a new OSD field shouldn't be controversial. We can always leave it off as a default and let the user choose. If there are no objections, I'd like to get this done before 1.7.3 is released or early in the 1.8 cycle.

However, I'd like to go further with this and make the dual indicator the default one, leaving the single purpose indicators available for users who might find them more useful. This would have the following advantages:

- Enabling the dual indicator instead of the current two ones by default would leave us more space for other indicators
- The dual indicator feels more polished (at least to me)
- Not strictly a requirement for this feature, but I'd also like to add support for displaying times bigger than 99 minutes and 59 seconds (the limit right now) by showing hours and minutes when the aforementioned limit is reached.

So, what are your thoughts on this?

fiam, 2017-09-06:
@digitalentity @DzikuVx Could we get your opinions on this one? Thanks!

digitalentity, 2017-09-08:
Let's deprecate "on time" indicator and replace it with "combined time". I think "fly time" indicator should stay regardless - for those who care only about time in the air.

krzysztofmatula, 2017-09-08:
> dual "on time" while unarmed / "fly time" while armed
Shouldn't it show "on time" only until first arm?
I mean, shouldn't "fly time" be also presented after disarming?

EnglishTurbines, 2017-09-09:
Arming the board sets the new home position and starts the "Fly Time" clock running.
After the flight is over, it would be good if it showed the total "Fly Time" in the Stats (because right now it doesnt it would seem)

Thats exactly whats required, **_speaking only as an end user_**....Not interested whatsoever in the "On Time"...it serves no purpose at all and just complicates matters.

fiam, 2017-09-15:
FC side merged, missing support on the configurator side.

fiam, 2017-09-15:
@EnglishTurbines Adding the fly time to the stats at the seems like a good idea. Just opened #2146 to track your request.

EnglishTurbines, 2017-09-16:
Great....please try and get the "On time" removed from iNAV.....It serves no purpose.......Less is more....:)

fiam, 2017-09-17:
@EnglishTurbines Both the combined time indicator and the fly time on the stats have been implemented. I've just merged support for the combined time indicator in the configurator, so I'm closing this one.

closed 2017-09-17 by fiam