iNavFlight/inav #2080: Servo/Gimbal Output not working

microlomaniac, 2017-09-07:
I'm on target NAZE. (flip32aio) I had this working in version 1.2.x I believe.

I want to output RX channel A4 to servo output 1, but nothing goes.
I enabled "Servo Gimbal" in the Configuration tab, which seem to work ok, as the motors are now mapped to outputs 3, 4, 5 and 6.
In the Servos tab, I enabled A4 for Servos 0 and 1.
In the "Motors" tab, when I turn the A4 knob on my transmitter, Servos 1 and 2 do reflect the change.

But the Servo I connected doesn't move.

Servo refresh rate is set to 50Hz.
It did work before on this exact board.
Same when enabling Channel forwarding. Doesn't change a thing.

dumps:

```
# status
System Uptime: 1035 seconds
Voltage: 128 * 0.1V (4S battery - CRITICAL)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, GPS=OK
Stack size: 2048, Stack address: 0x20005000
I2C Errors: 0, config size: 1420, max available config: 2048
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = ADC 1
  AIRSPEED : configured = none, used = none
System load: 16, cycle time: 2017, PID rate: 495, RX rate: 83, System rate: 9

# dump
# INAV/NAZE 1.7.2 Sep  6 2017 / 23:00:37 (81b85de)
mixer QUADX

mmix reset

servo 0 1000 2000 1500 100 7 
servo 1 1000 2000 1500 100 7 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 
smix reset

feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
[... 305 lines cut ...]
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

# version
# INAV/NAZE 1.7.2 Sep  6 2017 / 23:00:37 (81b85de)

#
```

digitalentity, 2017-09-08:
Can't reproduce neither on 1.7.2 nor on latest development. Servos connected to M1 and M2 correctly respond to AUX4.

DzikuVx, 2017-09-26:
Since no new data appeared, closing this one. In case of new data please reopen

closed 2017-09-26 by DzikuVx