iNavFlight/inav #2089: EZUHF.....SPPM Inputs....Are they supported on the F4 Boards?

EnglishTurbines, 2017-09-09:
Question is as Title, is this receiver sppm input supported on the F4 Boards.?????

 EZUHF is a popular FPV Control link, I think the RX output may be digital?
 Documentation seems limited about what receiver types are supported..?

digitalentity, 2017-09-09:
What's SPPM?

EnglishTurbines, 2017-09-09:
Shared Pulse Proportional Mode..?

EnglishTurbines, 2017-09-09:
Or "Muxed" output from the EZUHF reciever.

digitalentity, 2017-09-09:
Most of the INAV-capable boards support PPM input (which is quite often referred to as CPPM).

EnglishTurbines, 2017-09-09:
Yes, agreed CPPM...I have my EZUHF receiver configured for Muxed output, its bound and working / talking to my Transmitter..
   Question is, is it supported and what should my set up look like inside iNAV..???
Im already using iNAV and an L9R FrSky receiver, it works great, but is the EZUHF supported is my question?

digitalentity, 2017-09-09:
From FC perspective EzUHF is no different than L9R or any other receiver if it's talking PPM, S.Bus or other supported protocool. I don't own a EzUHF, but I know people are using it (not sure in S.Bus or PPM mode though)

EnglishTurbines, 2017-09-09:
Yes, thanks...it would be great if you could ask them how they have their port settings inside iNAV then?

 Can't seem to get receiver outputs going inside iNAV.....Help welcome please....:)

digitalentity, 2017-09-09:
What board are you using, how your receiver is connected and what settings do you have for receiver in INAV?

EnglishTurbines, 2017-09-09:
Hi....Yes, I am using an Omnibus F4 V1 Board.
 I have my EZUHF RX setup for "Muxed" or ppm output on CH1.
Inside iNAV, I have PPM enabled....(which I assume means SBUS is disabled)

  Thanks...:)

DzikuVx, 2017-09-09:
@EnglishTurbines if you have anything connected to UART1 that explains it. PPM input and UART1 are shared by default. 
This explains it or similar board https://quadmeup.com/uart1-and-ppm-on-airbot-f4-flip32-f4-flight-controllers/

EnglishTurbines, 2017-09-09:
Thanks, but the board in the Link photo is not the same as mine.
   The F4 (Omnibus) V1 board I have has what looks like two (open) jumper links.
    One for SBUS and One for PPM.
  Having substituted a FrSky RX, I can get RX Channel outputs fine, so the board is working ok. That leaves the question, does the EZUHF PPM output require the jumpers shorted????? and why would it require no jumpers shorted for SBUS?????

EnglishTurbines, 2017-09-09:
Happy Days......Removing the SBUS Jumper allowed the SPPM stream on the EZUHF receiver to work...:)

 Tip:
 Since I had the Servo Pinout Rail soldered on, I decided to try and lever off the jumper using a Scalpel...it worked......Using a soldering iron would have been really tricky.

 Thanks to those who helped me out with this...:)

DzikuVx, 2017-09-09:
Like I said, UART1 was the problem. Solved, closing

closed 2017-09-09 by DzikuVx