iNavFlight/inav #2107: SP Racing F4 Evo - flash iNav 1.73 ok, but no COM-port (unkown device)

Target0815, 2017-09-10:
Hi,

flash of iNav 1.73 in DFU mode on this board is ok, but after reboot there is no COM port (unknown device).

If I flash Cleanflight instead, everything's okay. COM port is assigned and the CF configurator can be used.

Is there perhaps a device ID missing?

digitalentity, 2017-09-11:
Hi! Just tested on my board - flashed with Cleanflight, then flashed with INAV 1.7.3 (SPRACINGF4EVO firmware). No issues - COM-port is detected correctly. Are you sure you are flashing the correct firmware?

Target0815, 2017-09-11:
Yes, I have flashed the correct firmware. I tested with two PCs (Win10). I also use both PCs to flash iNav to Omnibus F4 or Revolution F4. I noticed on the SP-Board that after flashing with Cleanflight this is displayed as Omnibus on one PC and as a Revolution on the other, so my question is, if there is a device ID or -identification missing.

I am still at work testing on another PC where there is no Cleanflight or iNav configuration installed. Let's see what is shown there when you plug in the board ...

digitalentity, 2017-09-11:
It uses standard ST's virtual COM-port drivers - same as Cleanflight and Betaflight - uses same Vendor ID and Device ID. I did test on Win7 and Linux thourh - not on Win10.

Target0815, 2017-09-11:
Yeah, I'm aware of that. Here at work it works immediately, so it must have to do with the drivers for the other boards. I think we can close the issue here and see if I can still find the problem.

Thank you!

digitalentity, 2017-09-11:
No worries. Please let us know if you figure it out on Win10 PCs - might be something specific about the drivers.

Target0815, 2017-09-17:
I took another look at the problem ...

In my opinion is the problem with the iNav configurator, because the problem occurs ONLY after iNav has flashed. If I flash once with iNav, I can't access the board via com-port anymore.

If I use Cleanflight, the Com-Port is always available after plugging in the board.

I noticed during the tests that Cleanflight 368 KB erases flash memory, while iNav only reports 256 KB erased memory. Is there something in the memory that causes this problem?

If I flash iNav as I said before, there is no com port available afterwards. If I now restart Windows 10, the com port is suddenly available and I can also use the iNav configurator. But only once, because when I unplug the board and then plug it in again, the com-port is gone again.

I use a Dell XPS 13 notebook with USB 3.0 ports. I have read about problems with F4 CPUs and USB 3.0, but the connection with iNav or Cleanflight Configurator is so clear that I think of a problem with the iNav configurator. Could it be that the flash routines of the CF Configurator are more up-to-date?

This speaks in favour of e. g. also that the CF Configurator directly detects the SPRACING F4 EVO board in DFU mode (no board selection necessary) while the iNav Configurator does not....

Target0815, 2017-09-17:
Addendum: I take the Betaflight Configurator it works like Cleanflight (delete 368 kB flash memory), with the difference that the board is NOT automatically detected in DFU mode like iNav. The Com-Port is always available even after unplugging or plugging in.

digitalentity, 2017-09-18:
Configurator erases as much memory as required to flash the firmware - no more.

Target0815, 2017-09-18:
I guess you're right.... When I flash the inav_1.7.3_SPRACINGF4EVO. hex with Cleanflight, only 256 kB of deleted memory is displayed. Still not running.

Is it because of the. hex file or is it the configurator or is my notebook a total waste? I mean, with CF or BF the board runs flawlessly, also with other boards (Omnibus, Matek, SP Racing F3) I have no problem...

Target0815, 2017-09-18:
Can you reopen the issue? Maybe another user has an idea?

Thanks!

markandkymward, 2017-09-20:
I've experienced the exact same issues as Target0815.  Nothing to add, just piling on.  Was hoping to clean up my 650 GPS build with this FC.

Target0815, 2017-09-22:
I have now found a solution for myself.... If a normal USB 2.0 hub is connected between the notebook's USB 3.0 port and the FC, the Com port is activated each time the FC is plugged in. The configurator will then also work.

But I had to try several USB hubs, it only worked with one of the tested ones. The real problem is the USB driver in iNav, which seems to be time critical. 

I tried to find differences between iNav-Cleanflight-Betaflight, but I didn't succeed without detailed knowledge of the architecture (I also program, but mostly database applications in the commercial area).

digitalentity, 2017-10-08:
Interesting finding. We don't seem to have the issue with other F4 boards - only with SPRacingF4 EVO. Maybe @hydra will chime in and shed some light if there's something specific about the F4 EVO's USB hardware?

Target0815, 2017-10-08:
As I said, I don't know the internals of iNav, but are there any differences between Cleanflight (it's always running) and iNav in the USB area?

markandkymward, 2017-10-13:
Has there been any movement on this?  I am still stuck

digitalentity, 2017-10-13:
Unfortunately I'm not able to reproduce the issue on board and all my PCs (linux, windows 7, windows 10). I can't tackle a bug which I can't reproduce.

markandkymward, 2017-10-14:
Discovered a work-around. If The usb is plugged into an unpowered FC the computer will not recognize it. If I apply 5v, then plug in the FC it connects straight away.  Auto connect in INav will not work in this case.  Works for both WIN10 and Linux.

matoy, 2018-01-01:
hi,

i have exactly same problem:
no problem with CF or BF,
can flash with inav but FC is not recognized after (no COM port),
tried by powering the board with 5V then connecting to usb: i can get inav configurator to work (@markandkymward : thanks for the workaround ! 👍 ),

i never experienced this kind of behavior with other boards.

there is clearly a problem with inav/configurator and this particular board.

hydra, 2018-01-02:
likely the cause is usb not being initialised early enough in system startup and/or being blocked by some other initialisation, likely mag.  this is why powering board via 5v and then connecting usb works .

spursmat, 2018-01-11:
Same issue here and same steps as above to supply 5V before connecting works.

Another workaround on Windows 10 is to uninstall the unrecognised device via device manager and then rescan for hardware changes. Device is magically recognised.

closed 2018-02-12 by digitalentity