iNavFlight/inav #2109: CC3D Change Mixer to something else then Quad-X is not working

DaPeace, 2017-09-11:
Ive tried to change the mixer to Hex-X on my CC3D but after saving and reboot its back on Quad X.
Ive tried to reflash but still the same. On Betaflight it works as expected..

Regards, Jan

oleost, 2017-09-11:
Hi.
Known bug, should be fixed in latest INAV 1.7.3 beta

DaPeace, 2017-09-11:
thx for the quick response. will try to flash 1.7.3 beta and see if it works..

digitalentity, 2017-09-11:
HEX-X issue is fixed in 1.7.3

DaPeace, 2017-09-11:
Not really.. ive tried out. Deleted the Flash, Flashed the hex-File from https://github.com/iNavFlight/inav/releases/tag/1.7.3

For me its the same. I cant even change to ANY other motor-scheme then quad-x on the CC3D.

digitalentity, 2017-09-11:
Being an F1 target CC3D only supports TRI, QUADP, QUADX, HEX6X multirotor motor layouts.

DaPeace, 2017-09-11:
Oh ok.. thank you.. why is betaflight working with hex-X?

digitalentity, 2017-09-11:
Because Betaflight is not INAV :smile: 
In INAV we removed support for "exotic" mixes on F1 targets for quite a while, otherwise CC3D and NAZE would be EOL'ed much earlier - see https://github.com/iNavFlight/inav/wiki/Custom-mixes-for-exotic-setups

DaPeace, 2017-09-11:
Thank you very much for your help. ive ordered  2 F3-Boards to play around.

closed 2017-10-09 by DzikuVx