iNavFlight/inav #2115: Naze32 build

tonyintn, 2017-09-11:
Having an issue building the latest release 1.7.3 beta  receiving the following errors
settings.yaml -> settings_generated.h, settings_generated.c
/bin/sh: ruby: command not found
make: *** [Makefile:933: src/main/fc/settings_generated.h] Error 127

Looks like the reference c and header files are missing from fc directory for setting_generated as they are being called but not in the directory.

Thanks

stronnag, 2017-09-11:
You need to install ruby. See the updated developer documentation.

tonyintn, 2017-09-11:
Thanks, for the quick response, suppose I should had look there to begin with my stupid mistake.

Thanks again

stronnag, 2017-09-11:
It should perhaps have been mentioned in the release note. OTOH, it's not unreasonable to assume folks building their own firmware will look in the PRs / git log for changes.

stronnag, 2017-09-11:
Added to release note. Closing.

closed 2017-09-11 by stronnag