iNavFlight/inav #2116: SP Racing F3 Deluxe - conflict when 2 softserials used simultaneously

Irish126, 2017-09-11:
Hi Guys,

I have my SP Racing F3 setup with iNav as follows with everything working fine.

Bluetooth on UART 1 (must disconnect to use USB) - OK
GPS at 38,400 on UART 2 - OK
SBUS on UART 3 - OK
Smart Port Telemetry on Softserial 2 (Tx and Rx shorted) - OK

Questions: 
1. Is it possible to add OSD on Softserial 1 with the above setup? My tests show that whenever anything is connected to Softserial 2 it disables any device on Softserial 1. 
2. Can BOTH softserial 1 AND 2 be used simultaneously? e.g. OSD (SS1) and Smartport Telemetry (SS2).
3. If both softserial devices are at 19,200 baud, I still can't get this working.
4. Is it true that COMBINED baud of 2 softserial must be max 19,200?
5. How to get 5 serial devices connected with SP Racing F3?  Must I switch from SBUS to PPM? Not ideal since I need more than 8 channels.

Trying to get this setup for an iNAV build with 5 serial devices.

Thank you.

andre92, 2017-09-12:
Smart port runs at 57.600 baud. Have you tried that baudrate for the OSD? I had GPS work at that baudrate in the past too, on softserial, combined with smart port.

Irish126, 2017-09-12:
Many thanks. Will give that a try and report back.

DzikuVx, 2017-09-12:
Unfortunately, 2 SS ports will fail if using bi-directional protocols. So, LTM + MSP will work, S.Port + MSP will fail. And it has nothing to do with speed.

One of the problems with SoftwareSerial I'm affraid

DzikuVx, 2017-10-06:
Also, it is not possible to run SoftwareSerials on different speeds. Closing, thanks

closed 2017-10-06 by DzikuVx