iNavFlight/inav #2117: OmnibusF4 Pro V3 on a Hexacopter

jstremmler, 2017-09-11:
If I try to configure iNav 1.7.3 for a X-hexacopter on my Omnibus F4 Pro V3 and want to use an additional LED strip, then this is impossible because, when activating the LED_strip in the config menu resource A1 is now reserved for the LED strip and no longer for motor 5. If I disable the feature for the LEDstrip in the config menu again then A1 is  correctly reserved for motor no. 5 of the hexacopter.

So here is the question: Why is the resource for the LED_strip not routed to B6 instead of A1?

digitalentity, 2017-09-12:
Good point. Omnibus V2 and above have a dedicated output for LEDs.

michaelvanderhoek, 2017-09-26:
Could this also be working for omnibus F4 Pro? (this is also called V2) Cant find port B6?

jstremmler, 2017-09-26:
Sure you can!

On the Omnibus F4 PRO board layout have a look at the bottom side. You will find there connector J1 with the 3 pinouts GND, 5V and LED_Strip. The pin for LED_Strip is connected to B6 of the processor and this is why iNAV should reserve LED_strip 1 as resource B6 instead of A1 which actually prevents you from using LED's and a hexacopter configuration same time.

michaelvanderhoek, 2017-09-26:
Hi jstremmler, 

Someone told me earlier that the dedicated led_strip on Omnibus F4 wasnt working at all. 
So what you are saying is that it IS possible. 

I'm happy to hear that! Already was going to sell my hexa because the ideas with LEDs weren't possible

jstremmler, 2017-09-26:
Hi Michael,

pls note that with iNav version 1.7.3 it is actually not possible to use a LED_Strip on a hexacopter with an Omnibus F4 Pro FC, but with version 1.8.0 it will be possible. If you wanted to use a LED_Strip on a hexacopter with iNav version 1.7.3 you would have to use a different FC. I think, e.g. an SPRACINGF3 would work.....

michaelvanderhoek, 2017-09-26:
Will wait for 1.8 then! Looking forward to it!

digitalentity, 2017-10-05:
@jstremmler can you test #2233 on your board? Please note that on cloned boards from banggood ledstrip connector is not usable)

jstremmler, 2017-10-07:
I tested it and the LED's work, but due to a thunderstorm and bad weather I did not test version 1.7.4 in flight. Best wishes to Komsomolsk! (Have visited it!)

michaelvanderhoek, 2017-10-07:
Hi jstremmler, what kind of board do you own? An original or clone?

jstremmler, 2017-10-08:
I own both, but as digitalentity poited out the clone cannot work because Chinese cloners connected the Led_Strip_Out pin of connector J1  on the board to a wrong pin of the processor.

digitalentity, 2017-10-08:
@jstremmler great! Then we can merge #2233. You should visit here again :wink:

michaelvanderhoek, 2017-10-08:
Jstremmler, oke tnx. I was just checking :D

closed 2017-10-08 by digitalentity

teckel12, 2017-10-11:
At least I read this and will know why my LEDs will stop working when I upgrade to 1.7.4.  I'd suggest we make a release notes (if this wasn't already thought of).

digitalentity, 2017-10-11:
@teckel12 good idea to mention that in release notes. There is a wiki work-in-progress page for that https://github.com/iNavFlight/inav/wiki/1.8.0-Release-notes

jstremmler, 2017-10-11:
Well, one couldn't  use LED strips with a hexacopter in iNav 1.7.3 before because A1 for motor 5 was resourced for the LED_Strips. 

To avoid confusion in iNAV version 1.7.4 I would like suggest that reserving B6 for the LED _Strip is only happening with a hexacopter frame! 
With any quadcopter resource A1 should be reserved. Then people with quadcopters (and Omnibus clones) and people with hexacopters (and an original Omnibus) can use LED_strips.

digitalentity, 2017-10-11:
@jstremmler I'd rather support LEDSTRIP as it is on original Airbot boards. Trying to support all of Chinese clones is impossible.

a6j, 2017-10-19:
If you do this:
> 
> define LED_STRIP
> if defined(OMNIBUSF4PRO) || defined(OMNIBUSF4V3)
>    define WS2811_PIN                      PA1
>    define WS2811_DMA_HANDLER_IDENTIFER    DMA1_ST4_HANDLER
>    define WS2811_DMA_STREAM               DMA1_Stream4
>    define WS2811_DMA_CHANNEL              DMA_Channel_6
> 
then LED STRIP will work on the OMNIBUSF4PRO clone with BG

putimir, 2017-10-25:
Hi, I have a cloned (BG) Omnibus F4 v2 (pro? - [prod. link](https://www.banggood.com/BF3_1_5-Omnibus-F4-V2-Flight-Controller-STM32-F405-MCU-Integrated-OSD-Built-in-5V-BEC-Current-Meter-p-1143258.html)), I connected the led strip to the J1 connector and I'm running iNav 1.8RC2, but the leds still aren't [working...](url)

My resource output says: 

`B05: LED1 OUT`
`B06: LED STRIP OUT`

So should it work or not? If it's true, that wiring is wrong, can I access the B06 from another pin?

teckel12, 2017-10-25:
With a clone it probably won't work on J1. You need to o use the new OMNIBUSF4PRO_LEDSTRIPM5 target and connect the LED strip to M5 (PWM5).

putimir, 2017-10-25:
I suppose this target isn't going to be pre-built, can I ask someone with a running dev environment to build it? Thank you!

teckel12, 2017-10-25:
It seems the new LEDSTEIPM5 target is not in the latest RC. It will be included in 1.8.

a6j, 2017-10-25:
[inav_1.8.0-RC2_OMNIBUSF4PRO_LEDSTRIPM5.hex](https://yadi.sk/d/8o27JTtK3P6heL)

simtcrom, 2018-10-30:
My FC is [Omnibus F4 Pro v2 (clone)](url

[image]
) and running inav 2.0.0 target OMNIBUSF4PRO_LEDSTRIPM5.
I am using PWM5 pin as data input to my LED strip.

I would like to do a DIY gimbal using two sg90 micro servos. As per the instructions, these servos to be connected to PWM5 and PWM6.

Can I simply remove the LED data input from PWM5 and connect servo there?
Or Do I have to reflash the FC with target OMNIBUSF4PRO?

vavisimo, 2019-09-08:
> With a clone it probably won't work on J1. You need to o use the new OMNIBUSF4PRO_LEDSTRIPM5 target and connect the LED strip to M5 (PWM5).

Hi Tim, may I ask, if clones are wrongly connected, did you find where did they connect PB6 to?
and if it is missleading and connected to nothing, the only way to use leds and M5 as PWM5 (needed for airplane, would be creating a tiny connection direct from chip leg PB6 to signal cable of led?

Thanks for reply ;)

PR #2233 (2017-10-08): Move ledstrip to PB6 on OBF4 PRO; Enable servo on M5/M6
Fixes #2117