iNavFlight/inav #2118: Blinking Current and Fly Mode fields on OSD in 1.7.3

Nihilistic, 2017-09-11:
Getting some blinking on my OSD with 1.7.3 on my Omnibus F3 in three fields. Amps, mAh used, and Fly Mode. Only the "mAh" label and the first digit blinks of that field.

digitalentity, 2017-09-16:
In post-1.7.3 OSD is going some serious overhaul. Let's see if this issue will remain in 1.8.

fiam, 2017-09-17:
@Nihilistic Can you post a screenshot? I think you might have aligned those fields in the artificial horizon area, because that would cause them to blink.

Nihilistic, 2017-09-17:
@fiam You're right. I had the current fields one line directly below where the sidebars end, and the fly mode  field directly above them. I moved the fly mode field to the left of the sidebar, and the current and mah fields down one line, and that resolved the issue. No more annoying blinking. Thanks man!

closed 2017-09-17 by Nihilistic

fiam, 2017-09-17:
@Nihilistic You're welcome :-). Just happened to me while I was implementing the new HDOP indicator. I hardcoded it to appear in the middle in the screen and it started blinking, that's when it hit me what you were probably having the same problem.

Nihilistic, 2017-09-17:
We're getting an HDOP indicator on the OSD? Sweeeet :)