iNavFlight/inav #2119: Default values for 600 sized multicopter

FlyingGustel, 2017-09-11:
https://github.com/iNavFlight/inav/wiki/Default-values-for-different-type-of-aircrafts

The inav wiki got already a textbox for 600 sized quadcopter but it doesn't contain any information yet.
I'm currently testing my tarrot 690 with an omnibus pro v3 flashed with inav 1.2 .
Did someone achieve experience with such a setup yet?

oleost, 2017-09-12:
Why are you use old outdated firmware?

oleost, 2017-09-12:
Also I'm closing this issue. Use rcgroups thread for support. But update your firmware first.

closed 2017-09-12 by oleost

FlyingGustel, 2017-09-12:
Oh sorry, that was by mistake.
I meant 1.7

FlyingGustel, 2017-09-12:
Why rcgroups? The inav wiki is simply not complete?

stronnag, 2017-09-12:
Anyone can edit the wiki, you are welcome to contribute.

oleost, 2017-09-12:
Then read the first line in that wiki page, in BOLD actually.

"This is outdated, INAV 1.6 is out and is using PRESET. Values here are old."

FlyingGustel, 2017-09-15:
Is it planned to start a new wiki?

stronnag, 2017-09-15:
Why would we do that? The existing wiki is a mass of useful information.

FlyingGustel, 2017-09-16:
No, just refering to the comment from oleost above mine. Not the whole wiki the default value page.😄