iNavFlight/inav #2120: Version 1.7.3 3D mode Arming Not Working Properly

KenImhof, 2017-09-12:
In version 1.7.3 the FC will not arm unless the throttle stick is at mid-throttle when arming occurs .  In all previous versions if the FC is armed at low throttle the motors will start to spin when center stick is reached.  In 1.7.3 if the FC is armed at low throttle they will not start to spin when mid-throttle is reached.

digitalentity, 2017-09-12:
Config dump?

KenImhof, 2017-09-12:
# dump

# version
# INAV/OMNIBUSF4V3 1.7.3 Sep  9 2017 / 10:37:25 (974440df)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature 3D
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
[... 457 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

KenImhof, 2017-09-17:
Appears that this issue is being ignored.  Seems to me that the person that removed the fc_core code and broke this feature in the last release should fix this problem.  I can certainly add the code back, but I am not sure why it was removed to being with.

fiam, 2017-09-17:
We really need a big cleanup of 3D related stuff (like making all NAV modes work with 3D mode). I plan to eventually address it, but got my plate full ATM with OSD related stuff.

KenImhof, 2017-09-17:
Hi Fiam,

That would be awesome.  Just let me know if I can help.  I can certainly do testing and I am in the process of coming up to speed with the iNav code base.

digitalentity, 2017-09-18:
@KenImhof the fact that issue is not being addressed immediately does not mean that it is ignored. Not many of us fly 3D mode. If you have found what's causing the bug - please share your findings.

digitalentity, 2017-09-18:
@fiam I'm planning a complete overhaul of `rcCommand`-related stuff to be independent of actual pwm values (for 1.9).
All of the code dealing with `rcCommand` will simply be requesting '75% normal thrust` or `20% reverse thrust`. We'll solve a lot of 3D-mode related issues in process.

KenImhof, 2017-10-31:
Submitted pull request #2433 as this problem also exists in 1.8 RC2.

digitalentity, 2017-11-01:
Let's define how arming in 3D mode should work.

KenImhof, 2017-11-01:
My opinion is that the switch arming operations should be:

#1. The FC should immediately arm when the arm switch is enabled and no disarm flags are set.

#2. If the arm switch is enabled and the throttle disarm flag is set the FC should arm when the throttle disarm flag is cleared provided that no other disarm flags are set.

#3. If the arm switch is enabled and all disarm flags are subsequently cleared and the throttle disarm flag is NOT the last flag cleared, the arm switch must be toggled to arm the FC.

digitalentity, 2017-11-01:
IMO case 2 is unsafe. Imagine this:
1. You flip the switch, machine doesn't arm. 
2. You forget to disable arming switch and go to pick up the quad. 
3. After you pick up the quad you accidentally move the throttle stich **through** the center position.
4. Quad arms and motors start spinning.

digitalentity, 2017-11-01:
I would keep current behavior (only allow arming if throttle is at zero). I guess the real issue here is finding "zero" throttle on a TX without self-centering throttle stick.

Maybe we should consider adding a feedback about zero throttle? Beeper, OSD, LED or something else?

KenImhof, 2017-11-01:
Case #2 is how it works in version 1.7.2 and all earlier versions.  This is also how it is works in betaflight, cleanflight, madcow, etc…

 

Since FC only arms when the throttle output is LOW, it should be save.  This is critical for 3d arming as LOW throttle is in the center so the typical 3d arming sequence is as follows:

 

1.        Throttle stick at the bottom (full reverse throttle).

2.        Enable arm switch.

3.        Move throttle stick to the center until the quad arms.

 

I also incorporated this operation into the last fc_core.c file that was send yesterday.

 

Thanks,

digitalentity, 2017-11-01:
1.7.2 didn't have the sophisticated pre-arming checks. We can simply remove the "sticky non-low throttle" flag for 3D mode.

KenImhof, 2017-11-01:
Submitted pull-request 2449.  I was able to simplify the code as I now under that the existing code to clear the ARMING_DISABLED_ARM_SWITCH flag so we can ARM even when the switch is in the disarmed position is okay as the switch arming is also checked at a higher level. The problem with just removing the  "sticky non-low throttle" flag for 3D mode is that the FC will arm with throttle high or as other disarm flags are cleared.

DzikuVx, 2020-04-14:
arming with Throttle in the middle will stay as arming method for Reversible Motors

closed 2020-04-14 by DzikuVx