iNavFlight/inav #2127: Attitude computation broken on latest development

DzikuVx, 2017-09-12:
Latest `development` d46272dcd42a51e83eb0664b39b4010287a6c99a

Attitude computation is broken when async gyro (gyro or all) is used. Computed attitude jump in all directions. 

It is fine when gyro async is disabled

stronnag, 2017-09-12:
I don't appear able to confirm this on multiple boards. Attitude is stable with `set async_mode = all`

Anything else I need to do?

DzikuVx, 2017-09-12:
Ok, maybe it is only F7. I will try to test different boards tomorrow

stronnag, 2017-09-12:
OK, seems stable on F3 and F4.

digitalentity, 2017-09-13:
@DzikuVx unable to confirm on neither REVO not SPRF3. Possibly an F7-related issue.

DzikuVx, 2017-09-13:
Tested on AnyFC F7 Pro and looks like there are more conditions for this bug to appear:
@stronnag @digitalentity 

Set:
1. Async gyro
1. Gyro LPF 256Hz
1. Flight controller looptime 500Hz

And then:
1. On Gyro frequency 500Hz and 1kHz -> works fine
1. On Gyro frequency 2kHz, 4kHz and 8kHz -> attitude is broken

stronnag, 2017-09-13:
For completeness, my test was with 1k gyro, matching your ok case.

DzikuVx, 2017-09-13:
Still, I had time to test only another F7, it mnight be fine on other CPUs

DzikuVx, 2017-09-13:
I somehow suspect this to be connected with

```
STATIC_FASTRAM float imuAccumulatedRate[XYZ_AXIS_COUNT];
STATIC_FASTRAM timeUs_t imuAccumulatedRateTimeUs;
STATIC_FASTRAM float imuAccumulatedAcc[XYZ_AXIS_COUNT];
STATIC_FASTRAM int   imuAccumulatedAccCount;
```

digitalentity, 2017-09-13:
@DzikuVx tested on REVO (F4) - no issues with any gyro sampling frequency in both `async = gyro` and `async = all`

DzikuVx, 2017-09-13:
Then we have another mystery... I will try to bisect and find a commit that broke it

DzikuVx, 2017-09-13:
@digitalentity no, it's not because of STATIC)FASTRAM in IMU. Staring to bisect

DzikuVx, 2017-09-13:
OK, looks like it hot broken at 3da0ab1862027bd177b132b385411b7413a28230

[image]

martinbudden, 2017-09-13:
One good reason to do migrations in small steps - it makes pinpointing the failure much easier.

PR #2133 (2017-09-14): Fix fast looptime attitude computation of F7 boards (probably more)
This apparently fixes #2127

@digitalentity @fiam @stronnag interesting discovery :)
Looks like FASTRAM and volatile do not mix well
closed 2017-09-14 by DzikuVx