iNavFlight/inav #2130: Consumed mAh in Telemetry not working on Omnibus F4 Pro v3

FPVZaphod, 2017-09-13:
Hello all,

In my Setup ( Omnibus F4 v3 Pro, Crossfire Micro v2 RX, Crossfire TX, Taranis, iNav 1.7.3 ) it shows the consumed mAh's in iNav GUI, but not in the Telemetry on the Taranis. The Sensor is recognized, but it remains 0. All others work fine.
Receiver is connected by UART1, CRSF Mode.

Thanks a lot - great work!

digitalentity, 2017-09-13:
Thanks for reporting. Will check. We might have a bug in CRSF telemetry code.

tgreer, 2017-09-16:
I have this exact config, so happy to do some test/debugging if required.

digitalentity, 2017-10-08:
@FPVZaphod @tgreer PR #2293 should fix the issue.

FPVZaphod, 2017-10-09:
Cool, thanks a lot!
How do I test this, any hex file available?

digitalentity, 2017-10-18:
@FPVZaphod can you please test if the issue is still present in recently released 1.8-RC1?

FPVZaphod, 2017-10-21:
Just tested it, works perfect on the Taranis now! Could not test it on the OSD yet, but this is independent from my Crossfire setup I guess. 
Thanks a lot!

closed 2017-10-21 by FPVZaphod