iNavFlight/inav #2135: inav 1.7.3 failed with ScitE Editor

ampere-dieter, 2017-09-14:
Hi all,
since inav 1.7.3 make with SciTE (V.4.0.0) failed with this message:
...
%% timer.c 
c:/users/wuertenberger/gnu_tools/build_tools/bin/sh: git: not found
settings.yaml -> settings_generated.h, settings_generated.c
c:/users/wuertenberger/gnu_tools/build_tools/bin/sh: ruby: not found
Makefile:932: recipe for target 'src/main/fc/settings_generated.h' failed
make: *** [src/main/fc/settings_generated.h] Error 127
>Exit code: 2

With inav 1.7.2 and all BetaFlight version I have no problems.

thanks for helps,
Dieter

digitalentity, 2017-09-14:
Duplicate of #2115

closed 2017-09-14 by digitalentity

stronnag, 2017-09-14:
You need to install ruby. See release note and developer documents.

ampere-dieter, 2017-09-14:
thanks stronnag for Your fast help.
With ruby installed all things are as usual...
Dieter