iNavFlight/inav #2141: Omnibus F4 osd电压不准

pdwdw, 2017-09-14:
Omnibus F4 osd显示电压不准,显示0.5v,调节也是这个范围变化,确定不是硬件问题,刷了几块都这样,刷bf
固件可以正常显示

digitalentity, 2017-09-14:
English please

fiam, 2017-09-14:
Just an automated translation, trying to make some sense of it:

```
Display voltage are not allowed to show 0.5v, the regulation is also the scope of change,
to determine the hardware is not a problem, brush a few are so, brush bf
Firmware can be displayed normally
```

So it seems to me that he can't get the decimal part of the voltage to show up somewhere (OSD? telemetry?) while in ßF it seems to work (`"brush bf
Firmware can be displayed normally"`).

Jozx900, 2017-09-17:
English:   
Omnibus F4 osd voltage could not correct display. Always show 0.5v. It's not Hardware problem, we try put Inav firmware to other F4, still the same. But F4 can show correct voltage when use BF firmware.

digitalentity, 2017-09-17:
My Omnibus F4 shows voltage correctly. For further troubleshooting a photo of the board and a CLI dump is required.

DzikuVx, 2017-10-06:
Since ticket seems to be stuck and seems like it is hardware connected, I'm closing it. Thanks

closed 2017-10-06 by DzikuVx