iNavFlight/inav #2150: Bad build Omnibus 1.7.3 building in windows light

Feldsalat, 2017-09-15:
The generated "inav_1.7.3_OMNIBUS.hex" seems to be ok until gps AND airplane is activated. When gps AND airplane is selected the communication with the board is no longer possible.  ->"Waiting for data"
Airplane without gps seems to works normal Quad X with gps also. 
I haven't made any changes to the firmware yet!
Downloaded firmware works fine but I need custom airplane with only one motor and three servos.
A self compiled inav 1.7.2 works without problems...

fiam, 2017-09-15:
@Feldsalat I've been investigating a weird hang on OMNIBUS F3 for a few days, but I haven't been able to reproduce it consistently. Could you post a dump of your configuration so we might try to reproduce the problem and fix it?

fiam, 2017-09-15:
Might be the same problem as in #2054

Feldsalat, 2017-09-15:
`# diff

# version
# INAV/OMNIBUS 1.7.3 Sep 12 2017 / 08:53:35 (974440d)

# resources

# mixer
mixer CUSTOMAIRPLANE
mmix 0  1.000  0.000  0.000  0.000

# servo

# servo mix
smix 0 3 0 100 0
smix 1 5 0 100 0
smix 2 2 1 100 0
smix 3 4 2 100 0

# feature
feature -RX_PPM
feature -CURRENT_METER
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature TELEMETRY
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper

# map
map TAER1234

# name

# serial
serial 0 32 115200 38400 0 115200
serial 1 2 115200 38400 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1350 2050
aux 1 2 0 1400 2075
aux 2 3 0 1575 2075
aux 3 3 2 1300 1700
aux 4 9 2 1300 1700
aux 5 8 2 1575 2075
aux 6 19 4 1300 2050
aux 7 21 6 1600 2050
aux 8 10 3 1300 2075
aux 9 14 5 1300 2050

# adjrange

# rxrange

# master
set gyro_hardware_lpf = 20HZ
set acc_hardware = MPU6000
set acczero_x = 184
set acczero_y = -28
set acczero_z = 128
set accgain_x = 4094
set accgain_y = 4088
set accgain_z = 4000
set mag_hardware = NONE
set mag_declination = 230
set baro_hardware = BMP085
set pitot_hardware = NONE
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_rate_num = 3
set blackbox_rate_denom = 4
set blackbox_device = SDCARD
set min_throttle = 1050
set max_throttle = 2000
set motor_pwm_rate = 50
set failsafe_delay = 30
set failsafe_procedure = RTH
set align_board_roll = 922
set align_board_pitch = 40
set vbat_min_cell_voltage = 30
set vbat_warning_cell_voltage = 32
set small_angle = 180
set gps_sbas_mode = EGNOS
set deadband = 6
set yaw_deadband = 6
set inav_use_gps_velned = OFF
set inav_reset_altitude = NEVER
set nav_extra_arming_safety = OFF
set nav_rth_allow_landing = OFF
set nav_rth_altitude = 2500
set nav_fw_cruise_thr = 1550
set nav_fw_min_thr = 1400
set nav_fw_max_thr = 1900
set nav_fw_bank_angle = 25
set nav_fw_loiter_radius = 4500
set nav_fw_launch_thr = 1800
set nav_fw_launch_idle_thr = 1400
set nav_fw_launch_climb_angle = 15
set osd_video_system = 1
set osd_units = METRIC
set osd_rssi_alarm = 40
set osd_time_alarm = 50
set osd_alt_alarm = 200
set osd_main_voltage_pos = 2530
set osd_rssi_pos = 2074
set osd_flytimer_pos = 2551
set osd_ontime_pos = 343
set osd_flymode_pos = 2081
set osd_throttle_pos = 2112
set osd_crosshairs = 2048
set osd_current_draw_pos = 97
set osd_mah_drawn_pos = 129
set osd_gps_speed_pos = 2274
set osd_gps_sats_pos = 3073
set osd_home_dir = 2477
set osd_home_dist = 2541
set osd_altitude_pos = 2295

# profile
profile 1

set fw_i_pitch = 15
set fw_ff_pitch = 90
set fw_p_roll = 15
set fw_i_roll = 15
set fw_p_yaw = 30
set fw_i_yaw = 5
set fw_ff_yaw = 75
set fw_p_level = 12
set max_angle_inclination_rll = 350
set rc_expo = 25
set rc_yaw_expo = 0
set roll_rate = 18
set pitch_rate = 10
set yaw_rate = 9
set tpa_rate = 20
set tpa_breakpoint = 1800`

fiam, 2017-09-15:
Thanks! I don't have an OMNIBUS here right now to test it (it's on a wing already loaded in the car for tomorrow morning), but I'll try to load your settings and see what happens over the weekend. I think @digitalentity might have an OMNIBUS handy, so he might be able to try it before me.

digitalentity, 2017-09-15:
@Feldsalat at what point it's getting stuck? Board is obviously detected, and you can click "Connect" button on Configurator. What tab are you accessing when Configurator hangs with "waiting for data" message?

Feldsalat, 2017-09-16:
Direct after save and reboot.

Schalonsus, 2017-09-16:
I can compile the hex without problems and it works correct for @Feldsalat 
We think this might be an issue with ruby (wrong installation or something)

Feldsalat, 2017-09-16:
I just followed this guide: [https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%20light.md](url)

[inav_1.7.3_OMNIBUS.zip](https://github.com/iNavFlight/inav/files/1308364/inav_1.7.3_OMNIBUS.zip)

Schalonsus, 2017-09-16:
What GCC Version are you running?

stronnag, 2017-09-16:
I fly an OMNIBUS regularly (like twice a week). No problems with 1.7.3 / 1.7.4 on either gcc 6.31, 7.1 or 7.2 (Arch Linux). So I'm surprised.

Feldsalat, 2017-09-16:
I now downloaded this version: [https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-win32.exe](url)
It now seems to work :-)

Schalonsus, 2017-09-16:
You can find the latest versions here: https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads

stronnag, 2017-09-16:
Those are by no means the latest version of  arm-none-eabi-gcc

Schalonsus, 2017-09-16:
my fault
https://gcc.gnu.org/mirrors.html

DzikuVx, 2017-10-06:
Looks like problem solved, I'm closing this one. Thanks

closed 2017-10-06 by DzikuVx

Feldsalat, 2017-10-07:
The instructions (Building in Windows light.md) still have to be updated! I do not know how...