iNavFlight/inav #2154: 1.7.4. MSP_STATUS bitfield implementation reverses boxflags field

stronnag, 2017-09-16:
I've been flying `de_rx_and_failsafe_cleanup` for the last few days, and noticed the following regression today (via development merge):

With 53114bf101a8ada57650afec3287b41f47639cd6 arming in HORIZON mode, I get logged by mwp
````
2017-09-16T16:32:42+0100 Masks arm 1 angle 2 horz 4 ph 200 rth 400 wp 800
2017-09-16T16:32:42+0100 switch val == 00040004
````  
The pertinent  bits (excuse the pun) being the first `horz 4` and the matching switch value.

Today, with f571000017da8df45d31c1e4b7b0bc78b87f4098, "Merge branch 'development' into de_rx_and_failsafe_cleanup":

````
2017-09-16T16:36:10+0100 Masks arm 1 angle 2 horz 4 ph 200 rth 400 wp 800
2017-09-16T16:36:10+0100 switch val == 20002000
````
If one reverses the bits, then the effect is clear, as this tacky little ruby script shows:
````
#!/usr/bin/ruby

hexval = 0x00040004
puts "%08x" % hexval
s= "%032b" % hexval
newhex = s.reverse.to_i(2)
puts "%08x" % newhex
````
thusly:
````
$ cd /tmp/
[jrh@eeyore]$ ./binvert.rb 
00040004
20002000
````
Please revert this behaviour; it affects 3rd party applications and breaks the historic MW/MSP principle that BOX masks 1,2,4 are respectively ARM, ANGLE and HORIZON.

fiam, 2017-09-16:
@stronnag This was definitely caused by #2142. I reversed the bit array implementation to be able to use the ASM instruction `clz` to iterate it. It seems the code for MSP_STATUS just dumps the bit array on the wire, so it now ends up in reverse. I'll submit a PR in a few minutes.

digitalentity, 2017-09-17:
Fixed

commit 86ec08fb (2017-09-17): Revert bit order in bitarray

Unfortunately, bitarrays are used in several places where the data
is finally written to the wire, so reversing their order would
mean breaking compatibility with software expecting those bits in
the reverse order.

This has the consequence to not allow us to use clz directly in
bitArrayFindFirstSet, requiring an intermediate rbit instruction
which adds around 1ns time penalty to max7456DrawScreenPartial().

Fixes #2154
closed 2017-09-17 by digitalentity