iNavFlight/inav #2155: Failsafe not detected on 1.7.3

marbalon, 2017-09-16:
Today I tried version 1.7.3 and catch this problem. Situation looks like below:

- sometimes when model lost RC link - RTH failsafe is detected correctly and model come back to home, clibm if needed
- when I turn off radio few times model turn back without any problem
- but when I tried to fly away to lost communication and force FS OSD shows RSSI 0%, no control and no FS :(, I increase transmit power, and get control again and go back to home.
- and last one I fly close to me but made aggressive flips, I lost RC link and model keeps last RC command and made rolls until hit the ground, but sadly in area when searching is not possible.

**Maybe this is related with eLeReS link only but never happened on version 1.7.1.**

So I don't have any logs, and any records because no SD card in gogles.
I will try to reproduce this on my desk, and catch it but just want to report because this is quite critical.
Model was Z84 wing with bad antennas but ideal to test FS, but found small place on the field when I can't go :(

marbalon, 2017-09-16:
Caught on my desk, time to time RSSI = 0, no control and not failsafe :(
[image]

stronnag, 2017-09-16:
As iNav monitors `rx_min_usec` for failsafe, would it not be more useful to show what that is when RSSI disappears ? Does it even check RSSI?

marbalon, 2017-09-16:
Ok, problem resolved (I hope).

closed 2017-09-16 by marbalon

skaman82, 2017-09-18:
I actually had the same problem on my desk. I have bound the crossfire micro v2 RX to my transmitter. Then during building I switched of my radio, but no failsafe was detected. I turned my radio on and off again and it worked like it should. Couldn't reproduce this glitch again but have not very much confidence right now because failsafe was definitely glitching one time.