iNavFlight/inav #2160: V1.7.3 : Matek F405-OSD/PDB Fixed wing - invalid mapping of Servo 4 to Output 2!

Pairan, 2017-09-17:
just wired up the Matek F405-OSD in combination with the Matek 6s HUB ... 

I am used to the following connections:
**SERVO 1** = ESC
**SERVO 3** = LEFT elevon
**SERVO 4** = RIGHT elevon

Now, SERVO 4 (alias S4) seems to be dead! Then I got some curious and tried S2 instead ... which turned out to be SERVO 4 !?

This then makes it as follows:
**SERVO 1** = ESC
**SERVO 3** = LEFT elevon
**SERVO 2** = RIGHT elevon

A mate has the same effect on his config/Matek F405-OSD !

the connections are made via the PDB. I tried to connect to the FC's S4 output but had the same effect.

@stronnag : if I remember right, you should have a Matek F405 at hands. can you confirm this?

digitalentity, 2017-09-17:
IIRC this is not wrong, this is what hardware allows. We need, however, to document this feature.

Pairan, 2017-09-17:
Ah ... i see . then we really need to document that on the MATEK Board MD so people can figure this!

digitalentity, 2017-09-17:
@Pairan I've took the liberty of renaming the issue to reflect the actual problem, I hope you don't mind :smile:

Pairan, 2017-09-17:
@digitalentity no worries - we figured "It's like that- and its not a bug!" then we surely can rename it to something more fitting :)

chtay, 2017-09-17:
Hi all
I encountered the same thing. But I have another problem. On my F3 board (SPRacing F3 Clone), I can configured customairplane which I use S3 to S6 for servo, but doing the same on this Matek F405-AIO, the S5 and S6 is not responding. If I set it to hexcopter, set motor signal as PWM, on motor test, the servos that connected to S5 and S6 is working.

Pairan, 2017-10-07:
I did a change to the documentation ... this could help, but maybe @stronnag will take a look at it

digitalentity, 2017-10-08:
#2269 is merged. Closing

closed 2017-10-08 by digitalentity