iNavFlight/inav #2165: bluetooth module @19200 not working since 1.7.3 beta

matoy, 2017-09-17:
Hi,

I have a HM10 bluetooth module connected to TX/RX UART1 on my omnibus f4 v2 pro.

same config with 1.7.2 and bluetooth is working.
on 1.7.3, no way to connect.

i don't know if that might be related to this on cf/bf:
https://github.com/betaflight/betaflight/issues/3967

but disabling OSD feature doesn't make work (it does on cf or bf).

here is a diff of the config:
https://pastebin.com/R48DcqNV

i agree that bluetooth is not very important on cf/bf but it is on inav to plan missions.

thanks to everybody for your hard work.

fiam, 2017-09-17:
@matoy I do have a few HM10 on the way, but they won't be here for a month or so. I do plan to install one of them in an OMNIBUSF4PRO, so if no one has fixed this one until then, I guess I'll take care of it :-).

digitalentity, 2017-09-18:
You are using S.Bus receiver on UART6 right? I might be able to test on my Omnibus F4 V3 PRO board and see if it exhibits the same behavior.

matoy, 2017-09-18:
Hi,

I have a ppm receiver (on ppm port obviously).
Same problem if I disable receiver in config and unplug it.

I have a v2 FC.

The very same config works on 1.7.2 and doesn't on 1.7.3

Thank you.

teckel12, 2017-09-18:
I have a bluetooth module I can test tonight.

ahfynjj, 2017-09-19:
我的蓝牙模块（hc-05 hc-06 hw-10）在ANYFC F7上使用正常，使用的波特率的57600

Hiddenbeach, 2017-09-20:
I am using Matek F405 FC. HM-10 Bluetooth module on com1. It works fine on Betaflight 3.2 RC5 but the same hardware setup doesn't work on iNav 1.7.3 and 1.7.4 (with software I2C). There is no Matek target prior 1.7.3.

digitalentity, 2017-09-20:
@Hiddenbeach please upload betaflight's working dump and INAV's non-working dump. Please do a "dump", not "diff"

digitalentity, 2017-09-20:
@Hiddenbeach can't reproduce on MATEK F405 OSD (INAV 1.7.3). Bluetooth is connected to UART1. MSP and telemetry works fine without issues.

digitalentity, 2017-09-20:
@matoy verified on Omnibus F4 V3. No issues.

Hiddenbeach, 2017-09-20:
dump files:
[bf_3.2_dump.txt](https://github.com/iNavFlight/inav/files/1316531/bf_3.2_dump.txt)
[inav_1.7.3_dump.txt](https://github.com/iNavFlight/inav/files/1316533/inav_1.7.3_dump.txt)

HM-10 on UART1 MSP, baud rate 9600. I just download bf 3.2 RC5 and iNav 1.7.3. everything on default setting except UART1. Anything else need to set?

digitalentity, 2017-09-20:
Still can't reproduce. All baud rates from 9600 to 115200 work with your dump and INAV 1.7.3

stronnag, 2017-09-20:
Matek f405 osd and AIO fine here with telemetry at 9600 baud. Don't think this is an FC or firmware issue.

digitalentity, 2017-09-20:
Changing from bug to support as this is most likely not related to board or firmware.

Hiddenbeach, 2017-09-21:
I tried MultiWii GUI and HM10 bluetooth serial lite. Both apps work with the setup. 

I think it's Mobile Flight related issue. It fails in the handshake process. Probably buffer overflow of HM-10 as described in this thread: https://www.rcgroups.com/forums/showthread.php?2601895-Mobile-Flight-Configuration-and-ground-control-app-for-Cleanflight-on-iPhone/page9

digitalentity, 2017-09-21:
@Hiddenbeach good, so it's a Mobileflight issue?

Hiddenbeach, 2017-09-21:
That's my assumption so far. Maybe we can make a local version of iNav with some features removed by commenting out some feature lines.

I've sent an email to the owner of Mobile Flight. He would be able to answer the question.

fiam, 2017-09-21:
1.7.3 has new boxes for modes related to the runcam split which take more space. If MobileFlight requests those on startup, it would explain why it worked on 1.7.2 but not on 1.7.3. Can you try a custom build without RC Split support?

matoy, 2017-09-23:
Hi,

you're right, i tried with 1.7.3 and an android device no problem to connect. And yes, i have connecting from an iphone and mobile flight in the first place.
sorry for disturbing.

thanks.

closed 2017-09-23 by matoy

matoy, 2017-09-23:
@Hiddenbeach : sorry but i don't understand your solution. i only have 3 features selected on inav (ledstrip, osd and blackbox) and i get the handshake failed.