iNavFlight/inav #2169: Allow negative value for 'current_meter_offset'

kingdraki, 2017-09-17:
Hi,

I'd kindly ask you to modify current_meter_offset to let user set up positive and negative values.

Here is the reason.
I'm using Omnibus F3 controller with Matek PDB. When the plane is idle, the current meter displays 0,10A but the measured value is 0,42A. This error continously presents (ie. when the measured current is 7,00A, the displayed current is around 6,70A, so I'm sure the scale is OK.), the current_meter_offset is 0 right now.
When I try to raise current_meter_offset, the displayed value will become lower, even negative values. So I should set up negative value for current_meter_offset itself for proper reading.

Thanks in advance,
regards Michael

DzikuVx, 2017-09-19:
@kingdraki take a look at https://github.com/iNavFlight/inav/pull/2176
But we will need someone to test it. Are you able to compile that branch by yourself?
I can compile for you, but I have to know the target

kingdraki, 2017-09-19:
Thanks for taking my suggestion into the code.
I have never compiled development branches before. Would appreciate if you
can compile for target OMNIBUS for testing.
Thanks.

DzikuVx, 2017-09-19:
@kingdraki please report if there are no problems
[inav_1.7.4_OMNIBUS.hex.zip](https://github.com/iNavFlight/inav/files/1315200/inav_1.7.4_OMNIBUS.hex.zip)

kingdraki, 2017-09-21:
Bench test was fine. I am able to finetune my current sensor now spot on.
Configurator did not let me enter values below zero at the moment.
CLI reports back allowed values -3300 to 3300 which is good.
Do you want me to do an in flight test too?

DzikuVx, 2017-09-22:
I suppose this is enought now, thanks

DzikuVx, 2017-09-22:
With pull request in place, closing this one

closed 2017-09-22 by DzikuVx