iNavFlight/inav #2172: INAV 1.73 (BETA) Build error 127

wolflonezx11, 2017-09-18:
Build error occurred.
I am looking for a solution.
Please

[/cygdrive/c/dev/inav]$ make TARGET=NAZE
settings.yaml -> settings_generated.h, settings_generated.c
/bin/sh: ruby: command not found
make: *** [Makefile:933: src/main/fc/settings_generated.h] Error 127
[/cygdrive/c/dev/inav]$

stronnag, 2017-09-18:
Install ruby
Read release notes an Dev docs
Search prior issues on this topic

DzikuVx, 2017-09-18:
@wolflonezx11 just like @stronnag wrote: you need ruby

closed 2017-09-18 by DzikuVx

wolflonezx11, 2017-09-20:
Install ruby

Become a problem solver Thank you so much

StefoTheHighElf, 2017-09-28:
For some reason I still get Ruby Problem:

/cygdrive/d/github/inav
$ make TARGET=NAZE
settings.yaml -> settings_generated.h, settings_generated.c
C:/Ruby24-x64/lib/ruby/2.4.0/open3.rb:199:in `spawn': No such file or directory - C:\Program Files (x86)\GNU Tools ARM Embedded\4.8 2014q2\bin/arm-none-eabi-g++.exe -mthumb -mcpu=cortex-m3 -flto -fuse-linker-plugin -Os -I./src/main -I./src/main/target -I./lib/main/STM32F10x_StdPeriph_Driver/inc -I./lib/main/CMSIS/CM3/CoreSupport -I./lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F10x -I./lib/main/MAVLink -I./src/main/target/NAZE -ggdb3 -DDEBUG -std=c++11 -Wall -Wextra -Wunsafe-loop-optimizations -Wdouble-promotion -ffunction-sections -fdata-sections -DSTM32F10X_MD -DSTM32F10X -DFLASH_SIZE=128 -DHSE_VALUE=8000000 -DUSE_STDPERIPH_DRIVER -DNAZE -pedantic -D__FORKNAME__=inav -D__TARGET__=NAZE -D__REVISION__=974440df -o obj/tmp/test obj/tmp/test.cpp (Errno::ENOENT)
        from C:/Ruby24-x64/lib/ruby/2.4.0/open3.rb:199:in `popen_run'
        from C:/Ruby24-x64/lib/ruby/2.4.0/open3.rb:95:in `popen3'
        from C:/Ruby24-x64/lib/ruby/2.4.0/open3.rb:258:in `capture3'
        from ./src/utils/settings.rb:647:in `block in compile_test_file'
        from ./src/utils/settings.rb:571:in `mktmpdir'
        from ./src/utils/settings.rb:623:in `compile_test_file'
        from ./src/utils/settings.rb:670:in `check_conditions'
        from ./src/utils/settings.rb:278:in `initialize'
        from ./src/utils/settings.rb:876:in `new'
        from ./src/utils/settings.rb:876:in `<main>'
make: *** [Makefile:933: src/main/fc/settings_generated.h] Error 1

Regards

klomfar, 2017-10-15:
Me too,how solve the problem?

digitalentity, 2017-10-15:
The problem is spaces in your PATH: `C:\Program Files (x86)\GNU Tools ARM Embedded\4.8 2014q2....`. Ruby on CYGWIN doesn't like spaces in PATH variable.

klomfar, 2017-10-15:
Thanks a lot but i have error with next line :/ Can you help me?
`$ make TARGET=NAZE
settings.yaml -> settings_generated.h, settings_generated.c
%% cli.c
./src/main/fc/cli.c:302:37: error: unknown type name 'clivalue_t'
 static void printValuePointer(const clivalue_t *var, const void *valuePointer, uint32_t full)
                                     ^
./src/main/fc/cli.c: In function 'printValuePointer':
./src/main/fc/cli.c:307:16: error: request for member 'type' in something not a structure or union
     switch (var->type & VALUE_TYPE_MASK) {
                ^
./src/main/fc/cli.c:307:25: error: 'VALUE_TYPE_MASK' undeclared (first use in this function)
     switch (var->type & VALUE_TYPE_MASK) {
                         ^
./src/main/fc/cli.c:307:25: note: each undeclared identifier is reported only once for each function it appears in
./src/main/fc/cli.c:308:10: error: 'VAR_UINT8' undeclared (first use in this function)
     case VAR_UINT8:
          ^
./src/main/fc/cli.c:312:10: error: 'VAR_INT8' undeclared (first use in this function)
     case VAR_INT8:
          ^
./src/main/fc/cli.c:316:10: error: 'VAR_UINT16' undeclared (first use in this function)
     case VAR_UINT16:
          ^
./src/main/fc/cli.c:320:10: error: 'VAR_INT16' undeclared (first use in this function)
     case VAR_INT16:
          ^
./src/main/fc/cli.c:324:10: error: 'VAR_UINT32' undeclared (first use in this function)
     case VAR_UINT32:
          ^
./src/main/fc/cli.c:328:10: error: 'VAR_FLOAT' undeclared (first use in this function)
     case VAR_FLOAT:
          ^
./src/main/fc/cli.c:331:21: error: request for member 'type' in something not a structure or union
             if ((var->type & VALUE_MODE_MASK) == MODE_DIRECT) {
                     ^
./src/main/fc/cli.c:331:30: error: 'VALUE_MODE_MASK' undeclared (first use in this function)
             if ((var->type & VALUE_MODE_MASK) == MODE_DIRECT) {
                              ^
./src/main/fc/cli.c:331:50: error: 'MODE_DIRECT' undeclared (first use in this function)
             if ((var->type & VALUE_MODE_MASK) == MODE_DIRECT) {
                                                  ^
./src/main/fc/cli.c:332:46: warning: implicit declaration of function 'clivalue_get_min' [-Wimplicit-function-declaration]
                 cliPrintf(" %s", ftoa((float)clivalue_get_min(var), buf));
                                              ^
./src/main/fc/cli.c:333:46: warning: implicit declaration of function 'clivalue_get_max' [-Wimplicit-function-declaration]
                 cliPrintf(" %s", ftoa((float)clivalue_get_max(var), buf));
                                              ^
./src/main/fc/cli.c:339:16: error: request for member 'type' in something not a structure or union
     switch (var->type & VALUE_MODE_MASK) {
                ^
./src/main/fc/cli.c:341:17: error: request for member 'type' in something not a structure or union
         if ((var->type & VALUE_TYPE_MASK) == VAR_UINT32)
                 ^
./src/main/fc/cli.c:346:21: error: request for member 'type' in something not a structure or union
             if ((var->type & VALUE_MODE_MASK) == MODE_DIRECT) {
                     ^
./src/main/fc/cli.c:351:10: error: 'MODE_LOOKUP' undeclared (first use in this function)
     case MODE_LOOKUP:
          ^
./src/main/fc/cli.c:352:16: error: request for member 'config' in something not a structure or union
         if (var->config.lookup.tableIndex < LOOKUP_TABLE_COUNT) {
                ^
./src/main/fc/cli.c:353:23: error: 'cliLookupTables' undeclared (first use in this function)
             cliPrintf(cliLookupTables[var->config.lookup.tableIndex].values[value]);
                       ^
./src/main/fc/cli.c:353:42: error: request for member 'config' in something not a structure or union
             cliPrintf(cliLookupTables[var->config.lookup.tableIndex].values[value]);
                                          ^
./src/main/fc/cli.c:355:13: warning: implicit declaration of function 'clivalue_get_name' [-Wimplicit-function-declaration]
[... 187 lines cut ...]
                                                                                  ^
./src/main/fc/cli.c:2266:87: error: 'VAR_UINT32' undeclared (first use in this function)
                                 else if ((cliValueTable[i].type & VALUE_TYPE_MASK) == VAR_UINT32)
                                                                                       ^
./src/main/fc/cli.c:2276:22: error: 'MODE_LOOKUP' undeclared (first use in this function)
                 case MODE_LOOKUP: {
                      ^
./src/main/fc/cli.c:2277:31: error: unknown type name 'lookupTableEntry_t'
                         const lookupTableEntry_t *tableEntry = &cliLookupTables[cliValueTable[i].config.lookup.tableIndex];
                               ^
./src/main/fc/cli.c:2277:65: error: 'cliLookupTables' undeclared (first use in this function)
                         const lookupTableEntry_t *tableEntry = &cliLookupTables[cliValueTable[i].config.lookup.tableIndex];
                                                                 ^
./src/main/fc/cli.c:2279:88: error: request for member 'valueCount' in something not a structure or union
                         for (uint32_t tableValueIndex = 0; tableValueIndex < tableEntry->valueCount && !matched; tableValueIndex++) {
                                                                                        ^
./src/main/fc/cli.c:2280:63: error: request for member 'values' in something not a structure or union
                             matched = sl_strcasecmp(tableEntry->values[tableValueIndex], eqptr) == 0;
                                                               ^
./src/main/fc/cli.c: In function 'printConfig':
./src/main/fc/cli.c:2599:23: error: 'MASTER_VALUE' undeclared (first use in this function)
         dumpAllValues(MASTER_VALUE, dumpMask);
                       ^
make: *** [Makefile:961: obj/main/NAZE/fc/cli.o] Chyba 1
`

stronnag, 2017-10-15:
a contemporary compiler might help, 4.8 is ancient.

fiam, 2017-10-15:
@klomfar You see to have your local copy in an inconsistent state. Remove:

- obj directory
- src/main/fc/settings_generated.h
- src/main/fc/settings_generated.c

Make sure your checkout is clean (check `git status` output). Also, as @stronnag has said an updated compiler might be required too. The settings generator wasn't tested with GCC 4.8.

klomfar, 2017-10-15:
thanks a lot, firmware compiled....

Arvycka, 2017-11-23:
/usr/bin/sh: ruby: command not found

Installed everything as described in https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows%20light.md

What Ruby version and where do I need to install?

Feldsalat, 2017-11-23:
https://github.com/iNavFlight/inav/pull/2280

Arvycka, 2017-11-23:
Thanks for the pointer. It worked for me.
However:

"##Install latest arm toolchain download https://gcc.gnu.org/mirrors.html"

This makes no sense and leads nowhere. Has to be:

"##Install latest arm toolchain download https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads"

Also:

"##Install Ruby
Install the latest Ruby version using Ruby Installer. Install directory MUST be "C:\devtools", or move files and folder from Ruby directory there manually"

These steps will fix this tutorial.

https://github.com/Schalonsus/inav/blob/cea385a24caac08a0ca48bb71b09ab687a979e7d/docs/development/Building%20in%20Windows%20light.md