iNavFlight/inav #2175: Issue using UART3 and PWM/SBUS

padcom, 2017-09-18:
Hi, I know this probably is the worst place to ask this kind of question but I am pretty much out of ideas what might be going on.

I have 2 of the Omnibus F4 V2 boards with SD Card slots and OSD. Up until recently one of those boards used to work just fine allowing me to use my PPM receiver and a BT converter.

I was soldering new set of wires to this board (it has an on-board current sensor) and after the operation both the PPM input as well as UART3 don't work.

I checked and double checked the connectivity between the actual BT board as well as the receiver. I have also verified on the other F4 board that the BT transciver as well as my RX do actually work.

I know there is a possibility that something went wrong during the soldering process (although I can hardly see how that would be possible since the soldering pads are gigantic and easy accessible and I an 100% sure I didn't touch any other electronics with my soldering iron).

Did something like that happen to anyone else? Is the UART on this F4 microcontroller really so fragile?

Does anyone has an idea how to check if the actual serial port is broken? Maybe some minimalist firmware doing echo on UART1 would be possible to check the well being of that port? 

If this isn't the proper place to be asking this sort of question can you direct me to a place that would be appropriate?

padcom, 2017-09-18:
https://www.banggood.com/BF3_1_5-Omnibus-F4-V2-Flight-Controller-STM32-F405-MCU-Integrated-OSD-Built-in-5V-BEC-Current-Meter-p-1143258.html

That's the exact board in question

DzikuVx, 2017-09-19:
It all depends on soldering iron, temperature and skills. It's not that easy to kill an uart, but if soldering iron in not ESD safe then it happens.
Are you sure it's not configuration? UART3 is shared with I2C so if you enable any I2C devide, UART3 will not work. And PPM is shared by default with UART1 so if you have something configured on UART1, PPM and UART1 would interfere/not work

closed 2017-09-19 by DzikuVx

padcom, 2017-10-06:
Ok, so it's not the ironing skills (although I too wouldn't rule it out in your place) but it is in fact a software issue. I have removed one of the 0 Ohm resistors leading to PPM/SBUS receiver and it all works now.

Does that help in any way? Is there anything that would require mechanical intervention to get the PPM and UART3 working after reflashing the board?