iNavFlight/inav #2180: omnibus f3 and gps issue

locutus40, 2017-09-20:
i have a omnibus f3 with https://www.banggood.com/UBLOX-NEO-M8N-BN-880-Flight-Control-GPS-Module-Dual-Module-Compass-p-971082.html?rmmds=myorder

this gps...

anyway im using inav..  new version with the latest firmware.. gps has a lock red light blinks i get 15 sats

its on a 250 frame.. but when i move the switch i set up to engage alt position hold and alt alttltude ... i mixed them on the inav so that it can  use one switch.. other switch is just horizon.. always on for stable flight.. anyway flies fine normally.. but when i flip over to gps nothing happens i mean it will dive alittle then again im 10 feet from me... maybe it needs ot be farther out before gps will work?

i lowered  the 500 cm mark on inav that you can use rth but i dont use rth i use land because i have a small yard.. anyway when i try it 10 feet out same thing it just does not work..

now i use a dji product or any other copter from the factory and it works i can let off the sticks and it hovers with no inputs...

i just cant figure out why its not working when it sees the sats... and theres really no more to set up./. i used mid throttle as the alt hold slider says... am i missing some setting?

im going bonkers here

teckel12, 2017-09-20:
Your description is a little confusing.  But, it sounds like maybe you need to set this to get the desired effect:
`set nav_use_midthr_for_althold = OFF`

Also, when you turn on your position and altitude hold (I like to call this 3D position hold), if it jumps a little you need to lower `nav_mc_hover_thr`.  If it drops a little you need to increase `nav_mc_hover_thr`.  My value for `nav_mc_hover_thr` is `1405` for a fairly typical 220mm frame with 2205 motors.  You may want to start there first.

In any case, when setup and calibrated correctly, when you switch it into 3D position hold, it just stops where it is in 3D space without touching the sticks (just like with DJI).  That is, as long as your throttle stick doesn't auto-center.

krzysztofmatula, 2017-09-20:
Shouldn't we aim at removing the "nav_mc_hover_thr" usage completely for baro-equipped machines?
It is perfectly doable to stay still without any predefined initial hover point. Inav is so stable on its hold positions and this small jump on engaging baro mode is only making bad impression...

digitalentity, 2017-09-20:
@krzysztofmatula you are misunderstanding the meaning of `nav_mc_hover_thr` - it's not the stick position. It's a hint for altitude hold code what throttle should be sent to ESCs for stable hover. If we won't have this parameter - we'll have to figure out something smart enough to figure it on the fly - and fast - because we may be arming and taking of in altitude hold mode.

krzysztofmatula, 2017-09-20:
> If we won't have this parameter - we'll have to figure out something smart enough to figure it on the fly - and fast - 
What about using current throttle (not the stick position, but the actual value) ?

digitalentity, 2017-09-20:
>What about using current throttle (not the stick position, but the actual value) ?
What throttle do we have at takeoff? Zero? Probably the worst possible estimate for hove throttle.

krzysztofmatula, 2017-09-20:
Hmm, this needs to be checked...
Actually, when you arm with baro mode enabled you don't expect the throttle to jump instantly to the hover value, do you? Maybe zero is not that bad?

digitalentity, 2017-09-20:
Yes, but the longer it will take for a quad to wind up the AltHold integrator - the worse will be overall takeoff experience. Same is true for i.e. doing a full throttle climb and hitting failsafe RTH. It will take quite a while for quad to figure out the true hover throttle - while it's doing so it will keep climbing at full throttle.

krzysztofmatula, 2017-09-20:
How long is "quite a while" ? If it is matter of single seconds I'd say it is acceptable.
If the error is big the correction should also be big (so: quick).

The proper manual setting of hover throttle is always problematic. People change batteries, attach/detach gimbals, and so on. In result the value is rarely good...

digitalentity, 2017-09-20:
@krzysztofmatula we have a small "jump" at the beginning of altitude hold because our initial guess (`nav_mc_hover_thr`) is close enough. Having less accurate guess merely makes the period of instability longer. And I don't think period of seconds is acceptable - imagine a "jump" lasting a whole second.

krzysztofmatula, 2017-09-20:
OK, I agree 1 second could be too long (imagined free fall with zero throttle and engaging alt-hold few meters above ground...)

But..., could we make using current throttle conditional, taken only when in hover position?
We can fine tune determining whether in hover position, but initially it could be zero-acceleration and zero-vertical speed. What do you think?

locutus40, 2017-09-20:
ill restate the issue.. the gps does not hold position and does not hold alititude.. it does nothing in fact when you flip the switch i set up.. and in inav you can see that the bars move to where i set up ie aux 2.... what i did was combine nav althold with nav alt poshold.. 

flip the switch and poof nothing in fact it goes crazy alittle and dives..... 

when im flipping the swtich im doing so 5 feet from me......

so what i was thinking is theres a tab that says 500cm away for rth to work....

so i lowered that to like 200 which is 6 feet from me i have yet to try it maybe thats it....

funny thing is i dont have rth checked i have land for the failsafe....

you go buy any quadcopter like dji and flip the gps button and its rock solid stays put 

even my 80 dollar hubsan h4 desire rock solid with 12 sats..

this is what i want on my omnibus..  but again when i flip the switch nothing..

then again i didnt try it with the lowered settings i will do so today and let you guy know....

but what i need is how to set up properly the gps in inav?"??????? hidden cli commands to fix issues stuff like that..

ill try to remove the hover at 1500 in the config tab thanks for that see if that does anything..

i used blue falcons a youtuber that passed away this year he did a series on how to hook up the gps to the omnibus f3 and that works i see 15 sats indoors i even see my house so i know its getting data...

its just the firmware 1.7.3 is not letting it hover at all rock solid it does nothing..

dont know how more clear i can be than that lol

you get a dji phantom and take off its rock solid just stays there why gps alt hold and position hold are always active..

with the inav firmwar you have to take off with gps off say with no mode on what i use is horizon when taking off.. then switch over to gps... poof nothing... this is what i need to fix  i see everyone elses working in youtube videos i want that lol

stronnag, 2017-09-20:
This is not a support channel, ranting here will not help. For the record, iNav works perfectly on my Omnibus / 250mm machine. PH, RTH, WP you name it --- perfect.

There is plenty of advice on the wiki about getting GPS and Mag to work. May I suggest you start there.

If you really want to be helped, go over to the RC Groups board, post a CLI diff and a blackbox log.

locutus40, 2017-09-20:
strong i dont appreciate being belittled like that.... whos ranting about anything im in need of help im explaining as best i can in detail what it does.. and if anyone knows how to fix it....

when i say dji stuff works out of the box etc what i mean is its probably the firmware or a issue with inav software..... i was using a referance i wasnt insulting the devs or anything..

look im to old and to tired to be arguing with someone... ive set up the gps and mag perfectly the gps works it shows sats so the omnibus shows its revieving the data ... calibrations are done....

ive gone though the calibration settings on inav.. theres one for use mid throttle for hover.. at 1500.. for the  alt altitude hold mode

ive combined altitude and pos hold mode on one channel.. ive tried seperating them and or just using one of them still gps does not come on or hod or  hold altitude....

there are no more boxes to check off or on in inav.. so im wondering if anyone is smarter than me :-) and can help;;; so far ive heard 1 person 1 give help the rest of you are just chatting off topic..

im not trying to be a jerk or anything....

im old i complain.. its in my nature you would too if you were 60 years old..

ok enough of that... can we just get back on topic.. anyway ill try the remove that thingie set nav_use_midthr_for_althold = OFF tekel12 suggested.. and see if that helps.. in cli  other than that i dont know what else to do.. i did set the rth distance to activate closer to me maybe im tooo close for gps to work im so used to it working on other quads from the home position... maybe it needs to be farther out i changed it to 200 or 300 so ill see later today or tomorrow and report back... i apreciate all the help... lets just remain civil

digitalentity, 2017-09-21:
@locutus40 please share a `diff` of your settings and output of `bootlog` command in CLI

locutus40, 2017-09-21:
ok im going to sound noobish... what is a diff? and bootlog and how do i get those from the interface?

stronnag, 2017-09-21:
In the CLI, `diff`` and  `bootlog`. Paste the output in code blocks here. At the risk of being a condescending 60 year old, some knowledge of the CLI is pretty much essential for trouble shooting. This is not hand-holding DJI.

locutus40, 2017-09-22:
heres the diff

PÉ
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/OMNIBUS 1.7.3 Sep  9 2017 / 10:09:29 (974440df)

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature RX_SERIAL
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER1234

# name

# serial
serial 0 0 115200 115200 0 115200
serial 1 2 115200 38400 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 1750 1900
aux 1 2 0 1100 1225
aux 2 3 1 1800 1900
aux 3 18 0 1800 1900

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set acc_hardware = MPU6000
set acczero_x = 102
set acczero_y = -61
set acczero_z = -536
set accgain_x = 4083
set accgain_y = 4099
set accgain_z = 4050
set mag_hardware = HMC5883
set magzero_x = 44
set magzero_y = -138
set magzero_z = -6
set baro_hardware = BMP280
set pitot_hardware = NONE
set blackbox_rate_denom = 8
set blackbox_device = SDCARD
set min_throttle = 1050
set max_throttle = 2000
set failsafe_throttle = 1400
set vbat_max_cell_voltage = 42
set vbat_warning_cell_voltage = 38
set gps_sbas_mode = AUTO
set nav_use_midthr_for_althold = ON
set nav_landing_speed = 150
set nav_land_slowdown_minalt = 300
set nav_land_slowdown_maxalt = 500
set nav_emerg_landing_speed = 450
set nav_min_rth_distance = 200
set nav_rth_climb_first = OFF
set nav_rth_abort_threshold = 250
set nav_rth_altitude = 200
set osd_cap_alarm = 850
set osd_time_alarm = 3
set osd_alt_alarm = 75
set osd_flymode_pos = 396
set osd_throttle_pos = 65
set osd_current_draw_pos = 97
set osd_gps_sats_pos = 352

# profile
profile 1

set mc_i_pitch = 50
set mc_d_pitch = 22
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
set roll_rate = 45
set pitch_rate = 45
set yaw_rate = 45

# 

heres the bootlog

# bootlog
Time Evt            Description  Parameters
  46:  0          CONFIG_LOADED 
  46:  1       SYSTEM_INIT_DONE 
 548: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 548: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 548: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 548: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 548: 18  TIMER_CHANNEL_SKIPPED  (5, 4, 0, 3)
 548: 18  TIMER_CHANNEL_SKIPPED  (6, 4, 0, 3)
 548:  2          PWM_INIT_DONE 
 559:  3       EXTRA_BOOT_DELAY 
1248:  9         GYRO_DETECTION  (6, 0, 0, 0)
1551: 10          ACC_DETECTION  (7, 0, 0, 0)
1571: 11         BARO_DETECTION  (4, 0, 0, 0)
1571: 20        PITOT_DETECTION  (0, 0, 0, 0)
1571: 12          MAG_DETECTION  (2, 0, 0, 0)
3205: 17    HMC5883L_SATURATION  (1, 0)
3306: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
3306:  4       SENSOR_INIT_DONE 
4009:  5          GPS_INIT_DONE 
5014:  8           SYSTEM_READY 

# 

let me know what i screwed up lol

teckel12, 2017-09-23:
Are
```
set nav_extra_arming_safety = ON
set gps_min_sats = 6
```
default settings? On my phone so I can't check. But maybe you don't have a GPS lock when you launch? It can take a few minutes at times.

locutus40, 2017-09-23:
yea there default settings .. i tried a few things removing the mid throttle for the gps just turning  it off.. when i flip my switch still does nothing i tried moving the copter farther out and flipping the switch nothing just hovers now.. ive tried just useing nav altposition hold and then just nav alt altitude still nothing.

its like non exsistant...

theres my new settings...Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/OMNIBUS 1.7.3 Sep  9 2017 / 10:09:29 (974440df)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature GPS
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
[... 468 lines cut ...]
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 45
set pitch_rate = 45
set yaw_rate = 45
set tpa_rate = 0
set tpa_breakpoint = 1500

# 

i wonder if i switch over to clean flight  does that have position hold and altititude hold?

i just dont get why this dont work i even recalibrated everything....

locutus40, 2017-09-25:
does anyone have any ideas at all? does clean flight have gps stuff like position hold and altitude hold? thats all i really want

DzikuVx, 2017-09-25:
I'm affraid Cleanflight does not do any GPS modes on OmnibusF3 any more. Or any other F3 board. And even when it was doing it, it was at least problematic.

You problem is happening only on your machine, I can confirm all GPS modes working 1.7.3 even with POSHOLD and ALTHOLD on the same switch. Maybe if you can record a blackbox log we would be able to help more

locutus40, 2017-09-25:
ok  clean flight is outta the ? i wanted to see if its the firmware.. see i bought this board a while back ill see if i can get another one sent to me.......
 ok ill do a black log box thing and send the file somehow on here? so you can read it.. ill do a hover flight then a few feet away ill engage the pos and alt hold.. maybe make a video of my setup etc i dumped 80 bucks into this thing so far i wanted a small 150 size gps :-( i suck at flying lol give me a few days to produce the file

locutus40, 2017-09-25:
Dziku vx can you do a video of it working? also i contacted banggood to see if they can send me a new f3 board

DzikuVx, 2017-09-25:
Video will not give an answer. Only blackbox log can

stronnag, 2017-09-25:
In addition to hover it would be useful to have some straight line flight so we can assess sensor behaviour against GPS and throttle.

locutus40, 2017-09-26:
i truly hate this copter... i go to get ready to record a few runs and nothing recorded to the memory stick so ill do it again tomorrow.. tried gps and what it does is wont let me move aliron or rudder or anything then dives to the left and loses altitude..... i think i have a faulty omnibus  bangood told me no on a resend.. im trying to convince them since i spent 20 bucks on this thing..... ill try again tomorrow to get it to record data.. it did 2 days ago now it wont .. i feel like tossing this thing out a window lol

digitalentity, 2017-09-26:
@locutus40 make sure you have SD-Card selected as your logging device in Configuration. Also make sure Blackbox feature is enabled in Configuration screen.

locutus40, 2017-09-26:
it is... i flew it last time it logged last time about 10 events..... im going to reflash the firmware and start over... i still think its going to do the same thing im begning to think its the board its self

locutus40, 2017-09-29:
ok finally got it to where it records.. dont know why it didnt.. but i reflashed everthing set up my quad.. dump is the same as last one... your gunna love this though..
set up my failsafe toggle and nave alt hold and altpos 
tested the hold/position same thing rises 2 feet then slowly drifts left and dives down i cant control it or revocover no matter what i do...
then youll laugh at this... i hit the failsafe set to land only with 1510 throttle as what it uses.. poof shot up 30 feet disapeared into the night then dove into my neighbors back yard i panicked never had this happen 830 at night knocked on there door broken english aka chinese oh i had some saplaining to do lol

anyway im finished with the gps stuff it isnt working it flies normally ok drifts right not gunna trim it with the transmitter but with acc function and fly it without gps...

but heres my logs... and pictures... i might sell it cheap to someone who wants it..6 bucks to ship in the usa only though.. ill use usps those ship anything that fits box... but i dunno... ill prolly include one battery the reciever is dsm2 and the flight board and 3d printed frame... dunno yet im thinking about it...

anyway heres the logs and pics of it... and btw i do have gps lock with 15 sats 3d lock too..

http://www.mediafire.com/file/pk3rj4084l2fdca/LOG00001.TXT
http://www.mediafire.com/file/91c402szvmc7x5t/LOG00002.TXT
http://www.mediafire.com/view/9meb27au8oqvvqj/20170921_195710.jpg
http://www.mediafire.com/view/gcwid45ffww3xw3/20170921_195716.jpg

flight times about 4 minutes same battery log 1-2 are battery never disconnected i just disarmed it picked it up after it dropped after 2 minutes hover after hitting the gps switch thats where you see a big disturbance in the gyro where it hit the ground the second one shows failsafe and hitting the ground ...

let me know guys suggestions.. im at a loss it should work..

i followed blue falcons tutorial the guy who died on his bike 2 months ago he shows how to wire one up... sca lca i think are all correct i get ordinates when i turn it and acell and gyro calibrated.. so im just scratching my head..

stronnag, 2017-09-29:
You might have a look at a video replay of the second log. 

https://vimeo.com/236121048

Wild movement in the artificial horizon, this is not a well set up machine; the firmware is defenceless in the face of this.

locutus40, 2017-09-29:
yea im noticing that too.. i think its the gyro so a main board replacement..
when i fly i have to hold right stick input and down its erratic because it wants to drift far to the left  and down so i have to input throttle up and down..... these  boards are clones from banggood... so im going to get a new one on repair status... i sent them that video to show them...

anyway how can i set it up bad the interface is all slider bars and check this off and that off... i calibrated the accell with a level a bubble level on all 6 axis so its perfectly level then the mag was calibrated.. then my radio set up switches etc.... i gave you my dump i didn't mess with that... set my mid throttle  and the fail-safe event was me testing it thats when it took a dive it should have recorded after that but it stopped.. so it sounds like its the board itself .....

btw how did you make that video what program did you use? a link would be cool..

btw guys and gals i appreciate the help.. i really do..

im so used to mission planner software and firmware... apm flight controllers and dji flight controllers the old stuff.. this new stuff im just getting used to.. im old so takes time...

anyway that software you used to check out the logs where can i get it? and that video display you used thats cool? wheres a link for that stuff ?

stronnag, 2017-09-29:
Well, it's quite obvious from the replay what when hovering (or attempting PH in the prior log), then the flying idea of level is not your bench idea of level. My Omnibus is an early Banggood offering and very stable; I think they went downhill on quality recently.

Software is mwp from mwptools https://github.com/stronnag/mwptools, https://www.rcgroups.com/forums/showthread.php?2633708-mwp, and a tacky little shell scipt to record a video.

locutus40, 2017-09-29:
i remember those cc3d flight controllers or the other ones being copied.. these are no diffrent clones upon clones etc.. i just sent them that video as proof... that its faulty hopefully they honor sending another one..

yea my idea of level flight is far from what its doing.. even taking off from a leveled surface it wants to fall left and down.. i have to fight the controls.. no matter how many times i calibrate the accell or mag it still does it... and when i do the gps hold it does the same thing.. cuts off all input aka i cant control it then it dives left and down then hits the ground.. when it should be holding position.

there was this flight  controller that had issues with a gyro a while back im talking early days cc3d or something like that well in fact i have one of those new around somewhere lol.. i also have a pixhawk micro.. i might switch over to it see if i can use it instead i know its old but hey maybe it will perform outta the box i dont know.. i choose the f3 because new technology faster etc ... so ill see what banggood says get a new one if that still has issues then i know its not the fc..

btw thanks for the info ill check it out

locutus40, 2017-09-29:
im looking for a exe file to run that little program cant find it in the github you sent me.. i know i know im such a noob this stuff is all new to me never heard of it seen it... i used to fly old school none of this fancy gps stuff and controllers etc.. so forgive me.. but how the heck do i take my log  and make it run like you did.. i downloaded the link... looked though all the files no exe to install a program or use it just a bunch of random files :-/

stronnag, 2017-09-29:
* It it linux / freebsd only
* You need to compile it yourself

locutus40, 2017-09-29:
i guess i have to research how to do that.. your speaking chinese to me lol

stronnag, 2017-09-29:
And without being (deliberately) rude or condescending, this is a technical software issue  channel; we have RC Groups and slack for banner. This issue should really be closed; it is not a firmware problem. Signing out.

locutus40, 2017-09-29:
stronnag and timhickle i think his name is have been nothing but rude and condecending to me.. what kinda commuinty is this anyway.. i came here for help.... it still might be software related.. geez i saw something cool aka software someone was using how else am i gunna sk where it came from.. on here of course...  you know ive been on forums before never before have i been treated liek this.... im closing this.. and going back to mission planner and dji products

closed 2017-09-29 by locutus40

digitalentity, 2017-10-01:
@locutus40 I concur with @stronnag - this is clearly not a software problem, but a hardware issue. Well known and (in)famous accelerometer resonance problem. Please soft-mount your FC.

locutus40, 2017-10-01:
ive closed this case..because 2 people were rude to me... to come to the conclusion it wasnt a software issue how would i know that unless i ask you people questions and see if anyone else has any idea... you all seem to think how dare i open a ticket because its a hardware issue but to get to know what it was i had to go this route.. infamouse? the only people that knew that is you people....  and so far only 1 of you the first guy to comment to me actually helped me out of all these comments he gave suggestions the rest of you were condescending and rude to me.. oh boy i went off topic asking for software i clearly thought might help me ohhh boy this and that..... dont mind me im old... i dont take #$%$ from no one lol  but your right i have it hard mounted because thats the only way i could do it.. ive seen others hard mount fc with no issues... what i think it is is a gyro or accell... since its a clone.. oh naze32 theres an original and clones thats what i was trying to remember naze32 controllers the clones had issues with the same hardware.... see but these f3 boards all you can get is clones or am i wrong?

digitalentity, 2017-10-01:
No one here tried to be rude to you here. If it looked like that to you, please don't take it too close. GitHub is mainly the place of more technical discussion - first line of community support usually happens in Slack channel or RCG thread. User community usually is able to figure out most of setup issues.

Regarding the issue - vibration issue is well known to the user community on RCG, someone would have resolved your issue much faster there. Usually self-leveling modes and acro are fine because they rely on accelerometer much less than navigation modes which are using mostly accelerometers to accurately calculate required corrections - this is why vibration is so harmful for INAV. Some get away with hard-mounted FCs - that's true, but usually happens on bigger frames and slow motors. On a racing drone soft-mounting is a must, otherwise you will get all sorts of issues with navigation modes.

Most boards out there are clones and lately their quality dropped like a stone. The problem is that you can't test quality of the board until you fly it in real-world environment - so it's impossible for firmware to detect possible issues before takeoff.

locutus40, 2017-10-01:
these 2 were condescending to me.. stronnag wait just the one guy i re read all the posts.. just that dude.....

more techinical discussion lol the tag says issues... click on it open a new issue this is a issue..... so this is where i posted to find people to help.. since people on youtube dont help and theres no 800 number with a tech behind it to talk too... lol  vibrations arent my issue.. the one user who was a pain in the butt to me.. showed me a flight log and when i asked how to get the software i was meet with scrorn how dare you go off topic lol  since its in a github its useless to me its not a exe file and i dont know how to compile and dont want to research 10 hours of videos and forums on how to make something work.. when mission planner works out of the box.. so im just going back to the good old days.. is all...

see these flight controllers arent ment for gps... they seem more for racing.. inav is just 90% copied from clean flight etc with some bits added in.. when clean flight should offer the whole thing anyway dont know why.... but it seems mission planner is far more advanced and theres no compiling etc just hoop up a gps to a apm or pixhawk and check off stuff and go i never had any real issues with that but i didnt want a 250 size drone or a full size drone i wanted a sub drone 150 or smaller with gps why battery prices i had a large drone 80 bucks per battery well hats expensive but a 150 size drone 10 bucks or 15 bucks for a 800 mah...  and yes everything now is clones cept naze and 3dr stuff and dji stuff... now the x4 desire from this one company it has gps and is small i could scrap that board and use it it flys like a dji no flipping switches to enter modes it just hovers but only problem it uses brused motors and the protocol is for brushed... .... so back to the pixhawmicro still works just gps takes for ever.. im going to swap the gps out for the one i have now since all the wires are the same maybe voltage is ill have to look

digitalentity, 2017-10-01:
Good luck to you putting APM or PixHawk on a 250 sized or smaller quad. All boards out there use same gyros and accelerometers, all boards are affected by vibration. Unless you solve your vibration issue I'm sure you'll end up in APM's issue tracker complaining about a fly away or inconsistent behavior.

locutus40, 2017-10-03:
lol who said it was a apm or pixhawk.. its a pixfalcon its 20mm x 20mm its as small as the setup with gps i have now...theres no vibration issue why do you keep saying that.. look at the gimble its going left and left and up and down aka crazy its a acceleromiter... you can hard mount any flight controller as long as your props are ballanced your frame is ridgid and motors ballanced... which mine is...

i have a quality issue i found that out though help from you guys.. turns out banggood blew me off wont refund me or replace it.. so i uploaded 3drs new firmware and im in the process of swapping out the omnibus.. no offence i ijust know 3dr stuff works out the box its older tech but its stable i even have a apm micro board from that one online supplier cant remember the uk based company expensive one ughh shipping with them sucks  any way... im going that route mission planner is point and click my micro setup even has telemery to a cell phone i printed out a cell phone holder and it shoots back maps at me just like dji even has a camera recieve for the solo.... no offence its not your guys firmware this is why i ask questions from you all i thought it was i didnt have the ability to see the gimple go nuts now i confirmed its a acell problem.. seeing how when i flip to gps it dives down and slowly to the left even fail safe youll notice that video at the end shoots up 20 feet which i programmed the hight 350 cm i think or 400.. and then took off left.. i even had to input right aliron  and pitch to keep it centered... but thanks for all the help... im not bashing inav firmware at all never was .... i just dont like being talked down to like im a idiot.. this is how i learn by asking questions and talking..

i still cant figure out how to compile that github to get what that guy used.. so :-/ anyway