iNavFlight/inav #2181: Drunken iNav RTH behaviour

konraddi, 2017-09-20:
Some unexplainable RTH issues happen here. Instead of flying straight back, my wing flies left and right, almost 90 degrees in the wrong direction. What could be the cause/solution?

Thanks,
Konrad

https://www.youtube.com/watch?v=DLJa5uz-axQ

# diff

# version
# INAV/SPRACINGF3 1.7.2 Jul 20 2017 / 11:45:09 (81b85de)

# resources

# mixer
mixer FLYING_WING

# servo
servo 3 1000 2000 1515 70 -1 
servo 4 1000 2000 1462 -70 -1 

# servo mix

# feature
feature -BLACKBOX
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper

# map
map RETA1234

# name

# serial
serial 1 2 115200 9600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1900 2100
aux 1 2 2 1900 2100
aux 2 20 1 900 2100
aux 3 8 3 1900 2100
aux 4 10 2 900 1075
aux 5 28 2 900 1125

# adjrange

# rxrange

# master
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set acc_hardware = MPU6050
set acczero_x = 217
set acczero_y = -40
set acczero_z = 209
set accgain_x = 4052
set accgain_y = 4072
set accgain_z = 3990
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 5
set serialrx_provider = SBUS
set min_throttle = 1200
set failsafe_procedure = RTH
set align_board_roll = -55
set align_board_yaw = 920
set disarm_kill_switch = OFF
set auto_disarm_delay = 0
set gps_sbas_mode = AUTO
set nav_rth_climb_first = OFF
set nav_rth_climb_ignore_emerg = ON
set nav_rth_alt_mode = MAX
set nav_fw_cruise_thr = 1350
set nav_fw_min_thr = 1150
set nav_fw_bank_angle = 30
set nav_fw_climb_angle = 35
set nav_fw_dive_angle = 20
set attitude_task_frequency = 500

# profile
profile 1

set fw_p_pitch = 4
set fw_i_pitch = 8
set fw_ff_pitch = 39
set fw_p_roll = 4
set fw_i_roll = 8
set fw_ff_roll = 36
set fw_p_yaw = 5
set fw_ff_yaw = 47
set fw_p_level = 15
set fw_i_level = 15
set fw_d_level = 50
set nav_fw_pos_z_p = 25
set nav_fw_pos_xy_p = 40
set nav_fw_pos_xy_i = 8
set nav_fw_pos_xy_d = 7
set rc_expo = 20
set roll_rate = 36
set pitch_rate = 15
set tpa_rate = 20

fiam, 2017-09-20:
@konraddi are you able to record a blackbox log of the wing while it's performing an RTH? That would really help diagnose the problem.

Schalonsus, 2017-09-21:
Your nav_fw_pos_xy_p is way too small. Try 75

fiveangle, 2017-09-22:
Try lower ABV batteries ? ;)

madoboy, 2017-10-06:
I had same issue yesterday in very windy conditions with my Mini Talon. I raised my bank angel to 30° before flying now i lowered back to 20°. I gonna test it again.

DzikuVx, 2017-10-11:
Since ticket seems to to stuck and at least one good advice was added, I'm closing it. Thank you

closed 2017-10-11 by DzikuVx

konraddi, 2017-10-13:
Hi @Schalonsus @DzikuVx,

many thanks for your advice. I raised nav_fw_pos_xy_p to 40 and then it worked fine.

My other wing, a Popwing 1300, when flying in RTH or Althold, does overshoot the horizon line and banks up and down several times before flying level. 
What values are the ones that apply for that? My Teksumo had those following values and he flew fine, in my Popwing diff there is none (because they are still untouched defaults?

set fw_p_level = 16
set fw_i_level = 15
set fw_d_level = 40

Are those the values that regulate the level flight in RTH and ALTHOLD (Angle/Horizon too)?

Many thanks,
Konrad

DzikuVx, 2017-10-13:
@konraddi yes, those values affects all flight modes besides Acro and Passthrough