iNavFlight/inav #2185: Propose alternative to LuaPilot lua script

teckel12, 2017-09-20:
I know iNav currently has https://github.com/iNavFlight/LuaPilot_Taranis_Telemetry.  But, it doesn't work on the Taranis Q X7 and is limited in some ways (at least for me).  I've created an alternative which also works on the Taranis Q X7 and provides many additional features that showcase the uniqueness of iNav and includes tools to help find a lost model and model orientation feedback.  The goal is to allow the lua script to be installed and with the minimum setup provide a full display and audio feedback of flight modes and status.

My current development is located here: https://github.com/teckel12/Taranis-iNav-Lua/

I'm wondering if making it part of the iNav repository would be useful to everyone.  I'll continue development in any case, but I can see it being very useful to more people than just myself.  I posted here to get feedback if this is a good idea or not.

BTW, this lua script is part of my learning process to add PID adjustment via S.Port telemetry like betaflight.  Part of which is code in the latest 1.7.3 build that I added to get ready to add PID adjustments to either this or a separate script.

fiam, 2017-09-20:
@teckel12 This looks very promising. I'm going to give it a bench test on my Q X7 before the week and hopefully several flight test during weekend!

teckel12, 2017-09-20:
The dial can be used to alternate between compass-based directional indicator and launch-based orientation and model location (two different display modes).  The launch-based orientation view shows the models's orientation based on launch orientation.  Also, if further than 25 feet away, it also shows the model's location in reference to launch location (the dotted line with an X at the end indicating the model location).  If the model would go down, you could use the model location direction and the distance to basically walk towards the downed model for recovery.

Once a flight is complete, you can also use the control dial to scroll between the current data values and the min/max values (an arrow up or down indicates that it's the max or the min value).

I really need to do a bit of screen documentation on what some of the less obvious things mean.  For example, the inverted arrow next to the altitude means alt hold.  Also, the round dot at the base of the orientation indicator means heading lock.  The rest should be fairly self-explanatory, but that's hard to judge as the author.

If you have logical switches and functions setup on your Taranis that play sounds, you'll probably want to make a copy of the model and delete all that before testing as the script gives audio feedback on all flight modes, navigation, hold status, high altitude, battery low, etc.

Anyway, I fly with this exclusively and it works very well.  But I'd love to get your feedback and using it to help with new iNav users.

digitalentity, 2017-09-21:
@teckel12 good stuff! I don't have a distinct preference for LuaPilot, but you should probably team up with @DzikuVx.

DzikuVx, 2017-09-21:
@teckel12 my work on LuaPilot for INAV is very scattery and all I was able to do is to display modes correctly. And I have no problem with scrapping that and overriding that with work from you.

@teckel12 Can you we found on our slack so we can talk about it more civilized way that comments?

teckel12, 2017-09-21:
@DzikuVx do I need an invite to join the iNav Slack workspace?

DzikuVx, 2017-09-22:
@teckel12 https://inavflight.signup.team/ you can invite yourself

teckel12, 2017-09-22:
@DzikuVx All good, on Slack now.

teckel12, 2017-09-23:
I believe this issue can be closed

https://github.com/iNavFlight/LuaTelemetry

fiam, 2017-09-25:
@teckel12 I agree. I'll just add 1.8 as the milestone to make sure we don't forget to mention it in the release notes and I'll close it.

closed 2017-09-25 by fiam

chief2811, 2018-11-18:
Hi, did you succesfully completed your work on inav lua script? i was looking for a PID Tuning script for inav. BF tx lua somehow worked with inav. But only for the standard PID's and not for the NavPID's.

teckel12, 2018-11-19:
@chief2811 This issue has nothing to do with PID adjustments. There's no INAV-specific PID adjustment Lua Script and I've never heard any developer ever talk about building one. It just doesn't seem important, and the BF one works for the core PID adjustment.

I also have no need for PID adjustment via transmitter, so I'm not the developer for this due to lack of interest.

What this issue is about has beed completed. It's called Lua Telemetry and available as part of the iNavFlight repo. But again, no PID adjustment nor will it ever be added as it's out of scope (and there's not enough transmitter memory to do it).