iNavFlight/inav #2189: Tweak artificial horizon indicator in OSD

giacomo892, 2017-09-21:
We should add the possibility to optionally transform the artificial horizon indicator into attitude indicator instead by basically reversing the actual displayed artificial horizon.

It will be nice as you immediately visually see how much your wing or multi is rolled.

fiam, 2017-09-28:
@giacomo892 I've been thinking about this one and it seems the problem to me is that we're lacking an aircraft reference in the indicator, which makes it more difficult to visually see your angle to the ground. I've just made this quick and dirty hack which shows a schematic airplane rather than a crosshair in the middle of the OSD. What do you think about it?

[image]
[image]

giacomo892, 2017-09-29:
@fiam Thanks for your precious work! It looks very neat, but do we really need the aircraft reference? Aren't the lateral arrows enough? I think this should provide enough reference for the purpose :) Like... "hey. i'm around 80 degrees now"

fiam, 2017-10-03:
Fixed in #2234

commit f52c9ada (2017-10-03): Add CLI setting osd_artificial_horizon_reverse_roll

Allows reversing the roll angle in the AHI indicator. Defaults to
OFF.

Fixes #2189
closed 2017-10-03 by fiam