iNavFlight/inav #2190: Omnibus F7 Pro  I2c mag remap possible ?

Blackcombify, 2017-09-21:
Hello Guys! This is my first Thread about Copters.

Ill tryed to build a 330 Copter with GSP Mag and Bluetooth. So i ordered the Omni F7 pro  and a Ublox GPS+Mag.

After i spent hours to get Inav flashed ill stuck on the MAG.
It seems as the Omni F7 pro has no I2C pads....

Is it possible to remap the pins ? to get a SDA/SCL?

I soldered my Spektrum Sat to uart1 i think, the GPS to uart3 and the Mag to uart6.  The Bluetooth part has no place anymore...

iam crying. Its my first autonom planned quad.

sry for my english, hope u understand.

greets

DzikuVx, 2017-09-22:
I2C is shared with UART3 on non-Pro Omnibus F7, so we can assume that it is the same in Pro version.
So Mag should go to I2C/UART3, GPS to UART1 and Sektrum to UART6.

closed 2017-09-22 by DzikuVx

Blackcombify, 2017-09-22:
Hi, thank u, ill try it . 5 minutes i give
a feedback

greets

Blackcombify, 2017-09-22:
you are the best man in universe

WORKS
i love you

DzikuVx, 2017-09-22:
Yea, I know...  :)

Blackcombify, 2017-09-22:
Ja Tiere an Option to get the hc06 Bluetooth in ?

Need 1 more uart ...right,?  Shit

⁣Gesendet mit BlueMail ​

DzikuVx, 2017-09-22:
You can try spektrum on PPM pin, this is also UART2 RX

Blackcombify, 2017-09-22:
Ok Ill try. Big thx

⁣Gesendet mit BlueMail ​

Blackcombify, 2017-09-22:
Think dont Work 

⁣Gesendet mit BlueMail ​

Blackcombify, 2017-09-22:
Hc06 in ppm ?

⁣Gesendet mit BlueMail ​

Blackcombify, 2017-09-22:
Only way is. Use RX TX for Bluetooth from USB
Like Plug an play. 

What do you think

⁣Gesendet mit BlueMail ​

alefesal, 2018-05-12:
Hello guys. Could someone tell me where uart 1 is in the omnibus f7 pro?

marcoskoller, 2023-06-15:
Hi everyone, any news?
I have an omnibus f7 pro that died the rcin. I saw that it is not possible to remap, can someone help me with pictures of the board? from the back and with the values ​​of 2 resistors?

At some point many years ago, I tried to connect a ppm receiver to this board, it didn't work at all. somewhere I read that I had to remove some resistors (at my own risk) and it went bad. I abandoned the saved sign, and now I have a project that I need to revive it. as I need INAV, I can't get soft serial, and I need the other two available uarts (i2c for compass and uart 6 for gps)

Below is the drive with pictures of the board (I don't know how to put pictures here). Realize that there is a pad missing from these resistors, I needed in addition to the value of the resistors, the mapping of this pad

[https://drive.google.com/drive/folders/1CttzKvTu8uta_DmnvkWzMgum4tC-oIcG?usp=sharing](url)