iNavFlight/inav #2193: No lumenier lux v2 firmware?

koetjeboe91, 2017-09-22:
Is it correct that there is currently no lux v2 support? had a naze 32 before the lux but that one died causing a crash.

digitalentity, 2017-09-24:
We don't have the hardware. If Lumenier will step in and donate something we can test - we can certainly add support for the board.

3sr3v3r, 2017-10-21:
Hi Koetjeboe91,

I have a micro lux v2 and have a fork of INAV where I included the LUXF4OSD target here:
https://github.com/3sr3v3r/inav
If you feel adventurous please test! I'm currently on holiday and haven't tested yet.
Will create a PR when I'm done testing.

uart3 can be used for I2C and the Led and S6 pads can be used as softserial.
Included the updated 'betaflight' soft serial implementation as well (currently a PR under review by jflyper) so that should solve the issue with the PA1 and PA8 pins being on different timers.

koetjeboe91, 2017-10-21:
Wil try and test it out wil try to leave some feedback later this week

3sr3v3r, 2017-10-21:
Cool. Put a compiled image here:
https://github.com/3sr3v3r/LUXF4OSD

digitalentity, 2017-10-22:
@3sr3v3r please submit a PR into INAV, we'll target it towards 1.9

koetjeboe91, 2017-10-22:
Have tryed to instal the hex file into the flash of the flight controller. It does flash but wil not boot (solid blue led). tryed to open a serial com to see what the processor is doing but it wont even show a com port. reflashing cleanflight or other software no problem.

3sr3v3r, 2017-10-22:
@koetjeboe91 thanks for the test!
Didn't bring my board on holiday so I will look into it when I'm back and when running create a PR.

Can't see an 'obvious' thing wrong at the moment.

3sr3v3r, 2017-11-07:
Right... returned home and no problems on on my Lux v2 micro. Created a pull request #2462.
Don't have a Lux V2 but after some research it turns out this board is completely different from Lux V2 micro.

Based on F3 a 64pin F3 MCU instead of F4.
Could be a nice board for INAV since it has 5 UARTs. 

This explains why the LUX V2 micro image wouldn't run on your board!

Can take a look what I can do but can't test since I don't have the board.

3sr3v3r, 2017-11-07:
@koetjeboe91 could you test this one?
Created a hex for the LUX V2.
You should be able to use UART2 for I2C (compass / baro)

[https://drive.google.com/drive/folders/1OeatggtOOroBXezK4SOQtUzLdyKvAw_r](url)

koetjeboe91, 2017-11-07:
wil try to flash it into the flight controller later this week need to wait
for some parts to fix my drone(tricopter).
wil also make a drone(quad) for the lux v2 where we can test software and
other stuff.

koetjeboe91, 2017-11-19:
oke update tryed to flash the lux v2 with the firmware and IT WORKS (yea).
Have not yet tryed to fly because i build a quad frame for it and has not
been nice weather outside so I cant tune the drone.
But motor spin and sensors looks correct. ps did not used the uarts yet
just got the baro in and orderd a new GPS  (
https://www.ebay.com/itm/Radiolink-M8N-GPS-Module-Ublox-M8030-for-Naze32-APM-CC3D-SP-F3-Naze32-Flip32-PX4/322725935682?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649
).
wil report back after first flight

2017-11-07 20:32 GMT+01:00 Peter Horstink <peterhorstink91@gmail.com>:

3sr3v3r, 2017-11-21:
@koetjeboe91 

From your nickname I suspect we don't live that far apart. Lots of koeien here and regen!
passhtrough bricked the esc? wow that doesn't sound good! Or were you able to 'revive' the esc by using cleanflight?

will compare the sources between INAV and Cleanflight. Have never touched that part of the code.

koetjeboe91, 2017-11-21:
de regen zou morgen minder worden maar ja nederland je weet maar nooit.
oke back on topic.
Yes I did revived the esc after I flashed cleanflight to the fc and then
use the pas trough of blhelisuite.
So there might be something wrong with that part of the firmware.
Sadly I did not yet reproduced the problem (dont realy want to).
So I can't say for 100% sure that it is the firmware.

3sr3v3r, 2017-11-22:
Good to hear you were able to revice the ESC's.
I actually didn't change anything from the basic Lux_Race board that I took a startingpoint (same as betaflight and cleanflight). 

After you've been able to test the serial ports / I2C port and it works I will create a pull request with the remark that blheli passthrough test wasn't successful (which esc board do you have?)

Could be that I have to create a new build with the compass drivers already included. They seem to be missing in this build.

koetjeboe91, 2017-11-25:
oke today I flown the drone for a hour or 2 without gps and baro (not yet
recieved) It flown great no strange things. hope to get the gps and baro
next week so I can test that out.
when this also works let me know what to test more.

koetjeboe91, 2017-12-09:
> @koetjeboe91 <https://github.com/koetjeboe91>
> So far so good I would say!
> For good Inav experience requires GPS, Baro and magnetometer.
>
> The radiolink GPS you referred to has a magnetometer build in.
>
>    - GPS can go to any free serial port (and configure this port for
>    ublox GPS)
>    - Magnetometer doesn't use serial port but I2C interface. You can
>    connect that to UART2 and not configure anything on the serial ports.
>
> Then the big question is what kind of baro you have.
> If it's I2C as well you can try to put that into the GPS box and connect
> to the SDA / SDL lines of the magnetometer. Can be on the same lines since
> both have a different address and that should work.

Oke Today I received the gps unit after installing it and setting it up
like normalI found that the gps takes a long time to lock.
Also the configuration tab does not want to store any sensor data or the
magnetometer declination after a reboot it goes back to none for
magnetometer (sensor) and 0 for the magnetometer declination.
On the Modes tab it does not show GPS home Hold or other
gps/baro/magnetometer related stuff.
Hope its a small thing.
but so far so good right?

3sr3v3r, 2017-12-17:
@koetjeboe91 
Only now I read your response.
Did you connect the magnetometer?
Normally if the board doesn't detect the magnetometer or baro it indeed will not store values.

Does your GPS unit come with 6 leads?

you could open the hex with a hex editor but it will not be very readable. It's the compiled firmware so you would need to set up a betaflight dev environment and have the source code.

koetjeboe91, 2017-12-17:
I wired the gps, baro and magneto meter the same as I did with a Matek f405
and that one does work.
If I then try to setup the lux v2 it does not want to work.

ps maybe its handy if we do some of the trouble shooting in dutch over pm
or other form of contact.

3sr3v3r, 2017-12-18:
@koetjeboe91

Indeed... don't know how to PM on GITHUB but I think I found you on rcgroups.
Sent you a PM there. My nick is DevJam.

3sr3v3r, 2018-01-07:
Update for this issue:
So far a NO GO for the LUX V2.
All is working except I2C interface.

UART2 (PA14 and PA15) are the only pins that should be able to function as I2C on this board but somehow it's not working. Connected my ST-Link debugger but since SWCLK also uses PA14 I can't debug and figure out what's wrong.

Dead end for me.
Did a search and the KFC32F3 target does seem to use the same pins for I2C on and F3 processor so that leaves me with some question marks.

If anybody has any idea please share. with 5 UARTS (but no OSD) the lux V2 could be an interesting board.

digitalentity, 2018-02-12:
I'm closing this since there is no activity towards resolving the issue. I can't assist since I don't have the hardware.

closed 2018-02-12 by digitalentity