iNavFlight/inav #2198: Softserials are instable in 1.7.3 on SPRACINGF3

mastersurferde, 2017-09-24:
Hi,
I have two boards SPRACINGF3 and I use Softserial 1 for Bluetooth and Softserial 2 for OSD.

Setup 1:
Receiver is connectes via PPM + SPORT on Serial 3 (X4RSB)
GPS on Serial 2

Setup 2:
Receiver is connected via TBS-Crossfire on Serial 3
GPS on Serial 2
Additional serial Connection via TBS-Crossfire on Serial 1

On both setups the Softserial Ports are unstable within some seconds. With Version 1.7.2 it is ok
My first Version was 1.7.1 - ok
Update to 1.7.3 - Softserial not ok
Downgrade to 1.7.2 - Softserial ok
Upgrade to 1.7.3 - Softserial not ok
Downgrade to 1.7.2 - Softserial ok.

Thank you for the projekt.
Best regards,
Stefan

Here is my dump:
# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 -100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
[... 423 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set tpa_rate = 33
set tpa_breakpoint = 1300

teckel12, 2017-09-25:
I'm only using one SS on the SP Racing Pro F3, but it's working just fine with 1.7.3 and it didn't work well with 1.7.2.  But, I'm using my SS port for S.Port telemetry.

Try selecting a baud rate instead of automatic.  Also, maybe a slower baud rate?  I'm running one SS port and selecting the baud rate specifically (57600) and it works without a hitch.

mastersurferde, 2017-09-25:
Hi teckel12,
thank you for your reply. I am using both ports in MSP mode with fixed 19200baud. This is exactly within the specification.
As I worte 1.7.2 is working fine - At the moment I have flashed 1.7.2 with no issues.
Regards,
Stefan

closed 2018-05-27 by stale[bot]