iNavFlight/inav #2202: Auto-detection of external BMP280 fails on iNav 1.7.4

Hiddenbeach, 2017-09-25:
Hi, I am using iNav 1.7.4 for my Matek F405 OSD build. There is an issue of external barometer BMP280. I know people have verified BMP280 works on iNav 1.7.4. Somehow my external BMP280 is auto-detected as MS5611. If I manually configure it as BMP280, iNav won't recognize it. Since iNav detects it as MS5611, it doesn't function properly thus runtime calibration fails.

Both external BMP280 and magnetometer (HMC5883 in GPS module Holybro M8n) are connected to same I2C port (UART3 of Matek F405 OSD). HMC5883 works fine.

I am using this external BMP280: https://www.amazon.com/gp/product/B01NAK6BL2/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

Did I miss something to configure? Anyone has this hardware configuration?

stronnag, 2017-09-25:
BMP280 is working just fine for me on MATEKF405OSD here (flown for several hours this weekend). There should be nothing to configure. Perhaps your baro is defective; can you try it on another FC?

sammichael25, 2017-10-04:
You should also note that your aircraft should be powered **before** you plug in your usb, else the baro wont be powered.

closed 2018-01-20 by DzikuVx