iNavFlight/inav #2205: SP Pro Racing F3 (Deluxe), baro and magnitometer is not working

retif, 2017-09-25:
FC page - https://flymod.net/item/flight_controller_sp_pro_racing_f3_deluxe
baro: MS5611
mag: HMC5883

versions tried: 1.7.3 1.7.2 1.7.1.

when i open configuration tab, select magnitometer HMC5883 and click save and reboot - mag icon becomes red and hardware health is red cross.

when i select baro MS5611 and save - baro icon is red, I2C errors start to grow and CPU Load is 666%, Serial Link icon is blinking, and i can not open any tab, its just "waiting for data", the only way reset this is to reflash FC with full erase

i tried to flash cleanflight v2.1.0 - maro and mag are working fine there

teckel12, 2017-09-25:
I have an SP Racing Pro F3 Deluxe and the barometer and magnetometer work just fine.  I'm running 1.7.3 now, but it also worked fine with earlier versions too.  It sounds like your looptime is set too high or your gyro sync is set to use too many CPU resources.  Try lowering your looptime and set gyro sync on and default aligned to gyro.  Or, post your CLI diff.

tonyintn, 2017-09-25:
Its because you are more then likely running a clone lots of them floating around now.

retif, 2017-09-25:
Yes, guys, i am using a clone. and Set i2c_speed = 200KHZ is does its job! now baro and mag are working. Thank you!

closed 2017-09-25 by retif